binsentsu/home-assistant-solaredge-modbus #130: StorEdge remote control reverts after setting registers

davidtmiles, 2022-12-24:
I have a two inverter (leader/follower via RS-485) setup. When setting Remote Control both inverts honor the request for about 10s before reverting to the Time of Use mode (which is set up in Monitoring). The changes are reflected in Storage Control Mode and Storage Defaul Mode but they seem to be "overriden". I have tried RRCR settings etc so unsure if its the control mechainsm, something about my setup or something on the monitoring portal (Admin > Energy Manager has storage profile unset (no check beside Profile set by installer) BUT i do see storage profiles still set on the inverter (though last update hasnt been for weeks). Do I need to set a profile in montoring to allow remote control to stay?

theneiljohnson, 2022-12-28:
Here's an example of a scene I use, which seems to work pretty well.

- id: '1671368899676'
  name: '[Batteries] Charge From Grid'
  entities:
    select.solaredge_storage_control_mode:
      options:
      - Disabled
      - Maximize Self Consumption
      - Time of Use
      - Backup Only
      - Remote Control
      friendly_name: solaredge (Storage Control Mode)
      state: Remote Control
    select.solaredge_storage_ac_charge_policy:
      options:
      - Disabled
      - Always Allowed
      - Fixed Energy Limit
      - Percent of Production
      friendly_name: solaredge (Storage AC Charge Policy)
      state: Always Allowed
    select.solaredge_storage_default_mode:
      options:
      - 'Off'
      - Charge from excess PV power only
      - Charge from PV first
      - Charge from PV and AC
      - Maximize export
      - Discharge to match load
      - Maximize self consumption
      friendly_name: solaredge (Storage Default Mode)
      state: Charge from PV and AC
    select.solaredge_storage_remote_command_mode:
      options:
      - 'Off'
      - Charge from excess PV power only
      - Charge from PV first
      - Charge from PV and AC
      - Maximize export
      - Discharge to match load
      - Maximize self consumption
      friendly_name: solaredge (Storage Remote Command Mode)
      state: Charge from PV and AC
    number.solaredge_storage_remote_command_timeout:
      min: 0
      max: 86400
      step: 1
      mode: auto
      unit_of_measurement: s
      friendly_name: solaredge (Storage Remote Command Timeout)
      state: '86400'
    select.solaredge_export_control_mode:
      options:
      - Disabled
      - Direct Export Limitation
      - Indirect Export Limitation
      - Production Limitation
      friendly_name: solaredge (Export control mode)
      state: Disabled

davidtmiles, 2022-12-28:
Thanks!

I found that in the Monitoring Portal, under Admin > Energy Manager.
Storage Profile  : Set by installer
Disable StorEdge

seems to have worked. Now I can set Remote Control via Modbus and not have settings revert.

closed 2022-12-28 by davidtmiles

mariov, 2023-10-04:
Similar issues here and tbh since the update nothing works not even through the app .. Installer has given me min admin access with only 3 tabs (Site details, Site access and revenue)

roger443, 2023-10-05:
Although not ideal, I have found a way to control the solaredge battery at times when I need to, despite the system wanting to overwrite ‘remote control’ and reverting to ‘maximize self consumption’ all the time.  It seems to revert about 10 seconds after selecting 'remote control’ in home assistant, so when I need control, I have home assistant automation set Storage Control Mode to ‘remote control’ every 9 seconds.  Then I can change things like Storage Default Mode to whatever I want (e.g. Charge from PV and AC) and it sticks!  So at least I can top up the battery on cloudy days from cheap grid power, before peak costs begin.  When I don’t need it anymore, I simply disable the automation, and it reverts backs to ‘maximize self consumption’.

I have a fault lodged with solaredge, but I don’t think it’s a very high priority, so not holding my breath.

mariov, 2023-10-05:
> Although not ideal, I have found a way to control the solaredge battery at times when I need to, despite the system wanting to overwrite ‘remote control’ and reverting to ‘maximize self consumption’ [...]

@roger443 Do you mind sharing how? I am very new to HA and when I try to create an automation I cant find any of the solaredge entities

theneiljohnson, 2023-10-05:
You need to disable StorEdge via SetApp, then it won't change back to MSC. Mine has been like this for a year or more and works great.

mariov, 2023-10-05:
SetApp said I have no permission . Solar Edge support has been useless and they said I have full admin. HA is my only option

theneiljohnson, 2023-10-05:
Maybe your installer can do it? You'll still need HA to control storage mode but once you have StorEdge disabled it won't keep trying to set it back to MSC.

roger443, 2023-10-06:
mariov, can you see any solaredge entities through Settings / Devices & Services / Entities?  If not, have you installed the HA solaredge modbus integration through HACS as described here https://github.com/binsentsu/home-assistant-solaredge-modbus and then enabled modbus over tcp?

mariov, 2023-10-06:
Yeah I can see 108 entities and I have already a dashboard but I don’t know how to set automation Sent from my iPhone

roger443, 2023-10-07:
You might need to look at some online help or YouTube for automation help, but below is my simple automation that keeps remote control active while it tops up the battery to a preset target level.  I have an input number helper that I can change to whatever target percentage I want, then I run the automation. It stops when the target is reached. I still need to manually set “charge from ac/pv” to actually initiate a manual charge.  Note, the automation never stops running if you set the target to 100% because obviously the battery charge can never be greater than that.

I had much more sophisticated automation running when it all worked ok. It would continuously monitor the state of charge and initiate charging at precisely the right times if it predicted solar wasn’t going to make.  The automation below is just a quick and dirty manual fix until Solaredge can make remote control stick.

Note: your entity names may vary ….

alias: "!Force Battery Charge 1"
description: ""
trigger: []
condition: []
action:
  - repeat:
      sequence:
        - service: select.select_option
          data:
            option: Remote Control
          target:
            entity_id: select.solaredge_storage_control_mode
        - delay:
            hours: 0
            minutes: 0
            seconds: 9
            milliseconds: 0
      while:
        - condition: numeric_state
          entity_id: sensor.solaredge_battery1_state_of_charge
          below: input_number.battery_target_charge
mode: single

roger443, 2023-10-07:
Thanks Neil. Unfortunately  I don’t have SetApp access and can’t see that option in setapp view mode or by browsing to the wifi hotspot on 172.16.0.1. I have asked both Solaredge and my installer to disable StorEdge. I’ll see what happens when/if they get around to it.  So frustrating that it ran really smoothly until a Solaredge software update killed remote control 

It would really help if you could tell me exactly what menu navigation they should use.  The more precise I can be with them, the better.

Thanks

dsmck58, 2023-10-12:
So fed up.  currently in an email exchange with Solaredge and they are just clueless. They keep telling me I can do everything using the TOU setting in the App.  They don't even know what their own App can do.