binsentsu/home-assistant-solaredge-modbus #132: Storage remote charge limit resets randomly

J1Buskens, 2023-01-04:
When I set the storage remote charge limit to 1kW (default 5kW) it remains there a few hours, but then at a random time reverts back to 5kW.

treynaer, 2023-01-04:
I'm seeing the same behavior.
I think we need a service that we can call to send the "Remote Storage settings" to the battery within the "Remote Command Timeout".

Another solution would be to periodically change the "Remote Charge Limit" with 1 watt difference within the "Remote Command Timeout". (1000W, 30 minutes later 1001W, 30 minutes later 1000W)

I have something similar for an Alfen Charger integrated via Modbus. To set the charge limit (via modbus write) I have to send the charge limit before the timer expires on the charger, otherwise the charger reverts to the safe value.

J1Buskens, 2023-01-05:
I checked the logs and found this at the same time:
```
2023-01-05 04:42:49.192 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 247, in read_modbus_data
    and self.read_modbus_data_meter1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 257, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 422, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 130, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)
```

J1Buskens, 2023-01-06:
> It's related to this #114

I was thinking the same, but when I changed it to >= the problem remains.
I guess the error and the resetting of the values are both results of the same cause.

ryanm101, 2023-01-30:
[image]

Are you sure it is not because of the Battery Remote Command Timeout?

ryanm101, 2023-01-30:
To charge my battery I do as i noted here: https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/104

the only difference is I now have an additional step to  set (Storage AC Charge Policy) to off / always allowed depending on time

ryanm101, 2023-01-30:
```yaml
alias: Charge Solar Battery From Grid
mode: single
trigger:
  - platform: state
    entity_id:
      - select.household_energy_daily
    to: offpeak
condition:
  - condition: and
    conditions:
      - condition: state
        entity_id: select.household_energy_daily
        state: offpeak
      - type: is_value
        condition: device
        device_id: 758241f74124df3fe7af56b6d84e1410
        entity_id: sensor.solaredge_battery1_state_of_charge
        domain: sensor
        below: 100
action:
  - service: select.select_option
    data:
      option: Charge from PV and AC
    target:
      entity_id: select.solaredge_storage_default_mode
  - service: select.select_option
    data:
      option: Always Allowed
    target:
      entity_id: select.solaredge_storage_ac_charge_policy
```

```yaml
alias: Charge Solar Battery From PV Only
description: ""
mode: single
trigger:
  - platform: state
    entity_id:
      - select.household_energy_daily
    to: peak
action:
  - service: select.select_option
    data:
      option: Maximize self consumption
    target:
      entity_id: select.solaredge_storage_default_mode
  - service: select.select_option
    data:
      option: Disabled
    target:
      entity_id: select.solaredge_storage_ac_charge_policy
```

```yaml
alias: Disable Solar Battery
description: ""
mode: single
trigger:
  - platform: numeric_state
    entity_id: sensor.solaredge_battery1_state_of_charge
    above: 99
condition:
  - condition: state
    entity_id: select.household_energy_daily
    state: offpeak
action:
  - service: select.select_option
    data:
      option: "Off"
    target:
      entity_id: select.solaredge_storage_default_mode
```

wayne-love, 2025-09-30:
Solaredge inverters will reset themselves twice a day.  Any settings not stored in flash memory (eg Storage Charge Limit) will revert to their defaults.  The reboot is why you are also seeing modbus timeouts at the same time.