binsentsu/home-assistant-solaredge-modbus #136: Integration broken with 2023.02.01

mxwi, 2023-02-02:
Hi,

the integration ist broken with 2023.02.01

```
2023-02-02 07:14:14.726 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solaredge_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-02-02 07:16:08.571 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.solaredge_modbus
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
2023-02-02 07:16:08.592 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration solaredge_modbus: Unable to import component: Exception importing custom_components.solaredge_modbus
2023-02-02 07:24:45.093 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solaredge_modbus.application_credentials
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
2023-02-02 07:24:48.188 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.solaredge_modbus
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
ImportError: Exception importing custom_components.solaredge_modbus
2023-02-02 07:24:56.424 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solaredge_modbus.application_credentials
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
2023-02-02 07:24:57.940 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.solaredge_modbus
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
ImportError: Exception importing custom_components.solaredge_modbus
2023-02-02 07:25:06.165 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solaredge_modbus.config_flow
File "/config/custom_components/solaredge_modbus/__init__.py", line 10, in <module>
2023-02-02 07:25:06.178 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solaredge_modbus: Exception importing custom_components.solaredge_modbus.config_flow
```

mxwi, 2023-02-02:
Guess it is more an issue with the modbus integration in HA in general.

closed 2023-02-02 by mxwi

binsentsu, 2023-02-02:
Hi. The modbus library bump to new major version was the reason. Issue should be resolved in release 1.8.0