binsentsu/home-assistant-solaredge-modbus #139: sensor.solaredge_ac_power in kW?

Marsman83, 2023-02-06:
The entity sensor.solaredge_ac_power shows the power in Watt.
With the new HA 2023.2.0 it should be possible to change the unit of measurement in the entity setting. However in my setup the option is not there, is that because there is a change needed in this addon?

For the entity sensor.solaredge_ac_energy_kwh the option to change the unit op measurement is there.

ChristophCaina, 2023-03-02:
The Power Sensors were only possible to change since the release of 2023.03.
Anyway - not for this integration, because they do not have the device class POWER yet.

This is something, the integration still needs to implement :)

ChristophCaina, 2023-03-02:
should be done with PR #147

ChristophCaina, 2023-03-10:
while the unit_of_measurement has been added (and the option should now be available) - I'Ve noticed, that the conversion of units is still not working on my system.

This might be related to some other changes that are still required (I've found another integration with the same issue so far)...Don't know, if this is an HA issue - but I guess, it is more likely something with this (and the other) integration... 

:-/