binsentsu/home-assistant-solaredge-modbus #143: cannot import name 'ModbusTcpClient' from 'pymodbus.client'

FileGo, 2023-02-13:
Integration version: v1.8.0
HA version: 2023.2.4

```
2023-02-13 17:30:21.756 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration solaredge_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-02-13 17:30:34.554 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration solaredge_modbus: Unable to import component: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)
```

taro1121, 2023-02-14:
Try "from pymodbus.client.sync import ModbusTcpClient".

FileGo, 2023-02-17:
> Couldn't reproduce your issue here. It suspect the pymodbus version which is used is still the old 1.5.2 version, which doesn't has the ModbusTcpClient class in pymodbus.client. Pymodbus version should be 3.1.1 from 1.8.0 onwards.

I think it might be the case of a conflict with another HACS extension I'm using ([givenergy-local](https://github.com/cdpuk/givenergy-local)), but it only becomes apparent if I upgrade this extension to 1.8.0 - it works fine with 1.7.2.

middelink, 2023-07-13:
I just restarted HA and have the same error. I have no other pymodbus clients to my knowledge.

```
Logger: homeassistant.setup
Source: custom_components/solax_modbus/__init__.py:31
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 09:37:20 (1 occurrences)
Last logged: 09:37:20

Setup failed for custom integration solax_modbus: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 13, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
```

middelink, 2023-07-13:
Sorry, ignore that. I did have a `modbus:` in my configuration.yaml. I removed it (it wasnt doing anything useful anyway) and the problem went away.