binsentsu/home-assistant-solaredge-modbus #148: Unit Conversion (W -> kW and Wh -> kWh -> MWh) is not working

ChristophCaina, 2023-03-10:
with the recent changes, HA does provide options to do a conversation of the Unit_of_measurements for some entities.
All Entities provided by this integration do show the Option to convert the Unit now, but the conversation itself does not work.

The sensor remains the same after applying a new unit_of_measurement.

The reason might be described somewhere in this article:

https://developers.home-assistant.io/blog/2022/12/07/unit_system_changes/

Example:

The Total (lifetime production) does provide its Unit in kWh.
[image]

I want to change the Unit to MWh...
[image]
[image]

after applying the update to the sensor, the Unit remains in kWh.
The same applies for W to kW changes....

I'm pretty sure that this has something to do with the Integration - I've found another Integration, with the same issue... other sensors / entities just worked perfectly fine.

ChristophCaina, 2023-05-30:
Hi @binsentsu - might I ask, if there's any progress on this?

binsentsu, 2023-05-30:
Yeah, in branch [feature/entityDescription_UpdateCoordinator](https://github.com/binsentsu/home-assistant-solaredge-modbus/tree/feature/entityDescription_UpdateCoordinator) I'm busy rewriting so all sensors use the HA core entityDescription model to bootstrap the sensors + datacoordinator updating mechanism. This way this functionality is transparantly added for this integration. It should already work and I will switch to it shortly to test run, but I'm only owning an inverter without meters or battery so there's always a chance of some regressions in that area.

ChristophCaina, 2023-05-30:
Well... I do have a Meter (but no battery) and can Join Beta Testing then (or now) :)

binsentsu, 2023-05-30:
thanks for testing!

ChristophCaina, 2023-06-03:
so far, I could not observe any issues.
There are some sensors, that do not report any values (0.0), but they are the same in my productive environment... So no issue with the changes so far :)

ChristophCaina, 2023-07-14:
Hi :)
Do you have an update about the state of this branch?

Hias80, 2023-12-18:
Hi,
When will this issue be resolved?

ChristophCaina, 2024-08-02:
Has this already be merged?
I haven't checked for quite a while now ... 😏🫣

KeyError34, 2026-09-04:
Subject: Bug report: Bar Chart component ignores system default_locale and user profile preferences in Grafana 13.2.1

Hello Team,We are facing an issue with number formatting (decimal separators) in the Bar Chart panel. We need to display numbers using the German locale format (comma as a decimal separator, e.g., 24,7 kWh instead of 24.7 kWh).

However, despite applying all correct server and user-level settings, the panel stubbornly renders numbers with American dots.

Our Environment & Steps Taken:Grafana Version: 

13.2.1 (Standalone binary installation under /usr/share/grafana/).

Server-side Config: Inside 
/usr/share/grafana/conf/defaults.ini, we have explicitly uncommented and set:[users] default_locale = de-DE[feature_toggles] localeFormatPreference = true

User Profile: Changed the "Language / Locale" to Deutsch (Deutschland) in the user preferences.

Panel Settings: 
Under the Bar Chart options, we tried setting the Einheit (Unit) to Lokales Format (Locale format), as well as keeping physical units (Watt-hour / kWh) with the configured system locale.

The Problem:The query returns raw numeric float values (e.g., 24703 Watt-hours). Grafana scales it to 24.7 kWh correctly, but it ignores the German locale completely on the chart's axes and labels, keeping the dot separator.

Our end-user needs to copy these values directly from the UI screen into Excel, but the dot separator forces them to manually find and replace dots with commas every time. Converting the SQL query output to a hardcoded string replace(..., '.', ',') is not an option because it breaks external automation scripts that fetch numeric data from this dashboard.

This behavior seems identical to the unresolved frontend internationalization bug discussed in GitHub Issue #116351.

Could you please clarify if there is a known frontend workaround for the Bar Chart component in version 13, or when a hotfix for this locale rendering bug is expected to be released?

Best regards,
Olha Lazyniuk ☺️