binsentsu/home-assistant-solaredge-modbus #149: Please add support for 3rd battery

Thebasestar, 2023-03-19:
Hi, can you please add support for 3rd battery? i own a setup with se6000h+3x se batterybank and i have no way to setup energy dashboard on HA cause of missing data from 3rd battery (calculated sensor etc )
My setup is all modbus connected (no energynet) and i can see values of all 3 batteries on SE cloud monitoring platform. (read somewhere 3rd battery datas are not available from modbus)

Many thanks for your great job

jovana, 2023-05-18:
You can check this more up to date repo, probably it has already support for the 3rd battery: https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/296

mpredfearn, 2023-05-18:
Nope, unfortunately this is a limitation of the (published) modbus register specification from Solaredge. There is no third battery in the specification and therefore it is impossible to support.

regardts-hub, 2024-01-04:
> Nope, unfortunately this is a limitation of the (published) modbus register specification from Solaredge. There is no third battery in the specification and therefore it is impossible to support.

The 3rd battery has an offset of +0x0200 on top of battery 2. I.e. Battery 3 State of Energy = 0xe484

Thebasestar, 2024-01-04:
> > Nope, unfortunately this is a limitation of the (published) modbus register specification from Solaredge. There is no third battery in the specification and therefore it is impossible to support.
> 
> The 3rd battery has an offset of +0x0200 on top of battery 2. I.e. Battery 3 State of Energy = 0xe484

Great, you made my day!!!!!
Lets hope binsetsu implement it soon.
Thanks again

Thebasestar, 2024-01-20:
Well for everybody needings, i made this modification by myself, and it work well adding battery 3 values to homeassistant.
I'm not an expert of github, i ve just made copy paste adding strings missing for 3rd battery, but i'd  like to share to all if possible,
@binsentsu i'm here if you want.

binsentsu, 2024-01-20:
If ready, you van make a pull request

Thebasestar, 2024-01-20:
Done, right i hope