binsentsu/home-assistant-solaredge-modbus #150: integration worked and then it didn't

bkbartk, 2023-03-21:
Hello,

I just installed this integration and all was working.
then I made a fatal error and needed to restore a my HA backup.

after restoring this backup I tried to install the integration again, but now nothing is working.
in the log are all timeouts as if it can't reach my inverter.

I see my inverter being connected to one of my accesspoints, but for some reason I can't connect

telnet tells me "connection failed"

I tried reconnecting to wifi.

disable and enable tcp modbus again.

but still the timeout remains where before restoring the backup all was working.

is there anything I am missing or need to reset?

ChristophCaina, 2023-03-22:
Hi, it might be, that the "modbus connection" was not closed properly after you run into the issue.
This might be the case why you had to wait several minutes.

Regarding the TimeOuts - personally, I would recommend a wired connection to your inverter, if possible.
We've also seen issues in the past, that SolarEdge blocked Wifi connections to the inverter in some cases (which was fixed in later firmware versions, but still... )

bkbartk, 2023-03-22:
I'm not sure, at the moment it's not stable for me
[image]
in the red squares, the horizontal lines there is no new data received.

unfortunately a wired connection is not possible. however I got the message about attaching an external wifi antenna.
so I'm plan on trying that first. unfortunately today I can't make it to the store.
if that doesn't fix the issue I will check how to update the firmware, but I couldn't even find which version I'm using right now.
It says CPU version 4.13.40
Version WSA 4.13.57
but I don't know it that's the FW or not.

bkbartk, 2023-03-24:
the external antenna seems to do the trick, I placed at and no hick ups at all for over 24h.

now I'm only worried about the IP67 classification as the smallest hole in the communication swivel was already used for gsm, So I needed to remove a little larger plug

bkbartk, 2023-04-01:
sorry to open the issue again,
the external antenna seemed to improve some stuff, however, after 9 days of working without any issues, the issue came back.

my log says:
```
This error originated from a custom integration.

Logger: pymodbus.client.tcp
Source: custom_components/solaredge_modbus/__init__.py:246
Integration: SolarEdge Modbus (documentation)
First occurred: March 30, 2023 at 21:57:12 (2589 occurrences)
Last logged: 20:04:50

Connection to (192.168.180.23, 1502) failed: timed out

---------------
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:246
Integration: SolarEdge Modbus (documentation)
First occurred: March 30, 2023 at 21:57:12 (2589 occurrences)
Last logged: 20:04:50

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 209, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 262, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 547, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
  File "/config/custom_components/solaredge_modbus/__init__.py", line 246, in read_holding_registers
    return self._client.read_holding_registers(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/mixin.py", line 104, in read_holding_registers
    return self.execute(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 194, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.180.23:1502)]

```
but at that moment I can ping my inverter.
releading the integration doesn't help, disabling for 5 minutes doesn't help either.

What helps is unpowering the inverter and start it again.

what I think,
and this is just a guess.
is that when the modbus connection is opened and something happens like a wifi glitch before the connection is closed,
the SolarEdge inverter waits indefinite until some close or finish command is received.

ChristophCaina, 2023-04-01:
OK. You might should Check Modbus Proxy.
Since a wired Connection is Not possible in your Case, I would Setup Modbus Proxy on a Raspberry Pi connected per LAN to your Inverter.

Then, Connect HomeAssistant to your Morbus Proxy via WiFi.
This might improve your Connection issues, since the Modbus Proxy keeps connected to the Inverter.

https://github.com/tiagocoutinho/modbus-proxy

bkbartk, 2023-04-02:
thanks, that would be a good solution. I maybe will try to contact solaredge first because I still think the issue is on their firmware, but this would be a great work around

closed 2023-04-02 by bkbartk

bkbartk, 2023-04-11:
FYI,
I contacted SolarEdge about this issue, and they told me Modbus TCP over WiFi is not supported anymore because of security issues. So I asked them about the details of those issues, but they(or at least this support employee) told me they didn't have details about this. 
Also he told me Modbus over TCP via WiFi is/will be completely disabled in the latest firmware. So then it will only work over Ethernet and RS485.

For now I changed the connection type to WiFi instead of automatic, So I an be sure my SolarEdge won't use the GSM modem by accident. And If I still have issue, I will try either the proxy or Ethernet over Power

ChristophCaina, 2023-04-11:
ok, thanks for reporting back.
IF SolarEdge does disable the ModbusConnection over WiFi, this will probably affect many users here...
So... personally, I would take the option to use ModbusProxy (connected per Ethernet) - and then use a Wifi Connection to the modbus proxy...
This might be a valid workaround for such a situation and we might consider to take this as an information to the readme?
@Binsentsu, what do you think?

bkbartk, 2023-04-13:
I required some additional information and they told me modbus over wifi is disabled since firmware version 4.14,
I'm now on v 4.13.57 (WSA?, I think) .
but I can't find anything about this, I see releasenots for V4.17 already. So I think a lot of users should already be affected by this one.
I requested a firmware update so I can confirm/decline that this is actually the case.
For now, I set up a wireless bridge to ethernet, Sorry not a modbus proxy as you suggested which actually is a way cooler solution, But I think it's harder to determine the problem if something breaks along the way.

bkbartk, 2023-04-14:
ok,
on V4.17.202 modbus over wifi still works, so if it ever was disabled in v4.14, it's enabled again.
not sure about the stability but I will find out.