binsentsu/home-assistant-solaredge-modbus #151: Wrong value M1 IMPORTED KWH ?

blowk, 2023-03-25:
Hi

The value of m1 imported kwh is not the sum of A B C.
Is this correct?

**Example**: 
```
(M1 IMPORTED A KWH)
1.145,688 kWh
(M1 IMPORTED B KWH)
0,0 kWh
(M1 IMPORTED C KWH)
19,47 kWh
(M1 IMPORTED KWH)
1.160,47 kWh
```

ChristophCaina, 2023-03-27:
it is the same behave for me:
[image]

As far as I can see, the Inverter is reporting the values for imported_a, imported_b, imported_c as well as the "imported" ... so it is right that the integration does not calculate this value...

But I don't know, where the difference is comming from.
Couldn't find this in the sunspec documentation ... :-(

blowk, 2023-09-07:
I've contacted SolarEdge and there seems to be a miscalculation. They didn't say if they will fix this, but the total value should be the right one and not the sum of a+b+c.

closed 2023-09-07 by blowk