binsentsu/home-assistant-solaredge-modbus #152: how to enable power control without battery?

rickroetenberg, 2023-03-29:
Hi,

How can I enable to power control function without a battery? It would be nice if I can change the power control (so the output to the grid) when energy pricing is negative.

Looking forward hearing from you.

Kind regards,

Rick

jtonk, 2023-04-19:
I'm also interested in this idea.

jovana, 2023-04-19:
> Hi,
> 
> 
> 
> How can I enable to power control function without a battery? It would be nice if I can change the power control (so the output to the grid) when energy pricing is negative.
> 
> 
> 
> Looking forward hearing from you.
> 
> 
> 
> Kind regards,
> 
> 
> 
> Rick

Hi,

I had asked the same question in here: https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/296

Johan-EU, 2023-05-21:
Hi,

Pull request #163 makes it possible to control the power output of an inverter. You don't need a battery.

binsentsu, 2023-05-30:
Released in [V1.9.0](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.9.0) Thanks for the effort @Johan-EU

closed 2023-05-30 by binsentsu