binsentsu/home-assistant-solaredge-modbus #154: Python Issue

beasthouse-au, 2023-04-20:
Hi, updated to HA 2023.4.5 today and now getting the following error when I go to restart HA.

`Platform error sensor.solaredge_local - Exception importing homeassistant.components.solaredge_local.sensor`

I looked into the logs and get the following.

```
2023-04-20 19:26:01.263 INFO (SyncWorker_6) [homeassistant.loader] Loaded solaredge_local from homeassistant.components.solaredge_local
2023-04-20 19:26:01.347 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform homeassistant.components.solaredge_local.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 779, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 796, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/solaredge_local/sensor.py", line 12, in <module>
    from solaredge_local import SolarEdge
  File "/usr/local/lib/python3.10/site-packages/solaredge_local/__init__.py", line 1, in <module>
    from solaredge_local.solaredge import SolarEdge
  File "/usr/local/lib/python3.10/site-packages/solaredge_local/solaredge.py", line 2, in <module>
    from .maintenance_pb2 import Maintenance
  File "/usr/local/lib/python3.10/site-packages/solaredge_local/maintenance_pb2.py", line 35, in <module>
    _descriptor.FieldDescriptor(
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/descriptor.py", line 561, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
2023-04-20 19:26:01.350 ERROR (MainThread) [homeassistant.config] Platform error: sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 779, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 796, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/solaredge_local/sensor.py", line 12, in <module>
    from solaredge_local import SolarEdge
  File "/usr/local/lib/python3.10/site-packages/solaredge_local/__init__.py", line 1, in <module>
    from solaredge_local.solaredge import SolarEdge
  File "/usr/local/lib/python3.10/site-packages/solaredge_local/solaredge.py", line 2, in <module>
    from .maintenance_pb2 import Maintenance
  File "/usr/local/lib/python3.10/site-packages/solaredge_local/maintenance_pb2.py", line 35, in <module>
    _descriptor.FieldDescriptor(
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/descriptor.py", line 561, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config.py", line 928, in async_process_component_config
    platform = p_integration.get_platform(domain)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 788, in get_platform
    raise ImportError(
ImportError: Exception importing homeassistant.components.solaredge_local.sensor
```

Any idea how to resolve this? The integration appears to be working still.

binsentsu, 2023-04-22:
This error comes from solaredge_local integration, not solaredge_modbus

closed 2023-04-22 by binsentsu

beasthouse-au, 2023-04-22:
> This error comes from solaredge_local integration, not solaredge_modbus

Oh wow - silly me 😂