binsentsu/home-assistant-solaredge-modbus #155: Is it possible to change polling frequency after initial setup?

Landcross, 2023-04-21:
I've been using this addon for quite some time now and I have no issues with it; it works like a charm. However, when I initially set up the integration I have set the polling frequency to 30s which at the time was fine for me. However, now I would like to decrease (or increase, depending on how you look at it) the frequency to e.g. 10s.

I checked if there are any editable integration settings in HA, but I couldn't find anything. Is it possible to change the frequency without removing and re-adding the integration?

Landcross, 2023-04-21:
I did some reading into the docs and it seems like to make this possible the addon needs to implement the Options Flow, as documented here: https://developers.home-assistant.io/docs/config_entries_options_flow_handler/

cyberax17, 2024-07-20:
This would be a great feature if implemented