binsentsu/home-assistant-solaredge-modbus #160: V1.8.0: No Data displayed in HA despite proper communication between HA and SE10K Inverter

ItsMe00007, 2023-05-11:
Hi,
I just setup a new SE10K (no optimisers/panels attached yet) with current firmware and enabled modbus tcp and verified it is working via "modbus tcp client" on my MacBook by successfully reading the model/serial via modbus tcp from the inverter..

But the HA does not display any data in the modbus tcp integration.
Therefore I installed TCPDUMP on HA and traced the modbus requests/replys from HA.
The HA integration connects properly to port 1502 of the inverter and requests/receives the following modbus data (extracted via Wireshark):
Request: 00010000000601039c870026
Reply:   00010000004f01034c0000000000000000fffe0fec0fcb0fd70932091f0924ffff000000001386fffe000000000000000000000000000000000000000000000002ffff00000000800006ae80008000fffe00010000

Request: 00020000000601039c870026
Reply:   00020000004f01034c0000000000000000fffe0fef0fd50fda0934092a0922ffff000000001386fffe000000000000000000000000000000000000000000000002ffff00000000800006ae80008000fffe00010000 

But there is no data displayed in the integration and it seems the integration has problems parsing the received data according to the log:
Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:143 
Integration: SolarEdge Modbus (documentation) 
First occurred: 17:54:31 (1 occurrences) 
Last logged: 17:54:31

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 209, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 262, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 626, in read_modbus_data_inverter
    acenergy = validate(self.calculate_value(acenergy, acenergysf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 143, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

ItsMe00007, 2023-05-11:
It seems that the problem is that acenergy is still 0 (newly installed and panels/optimizers will arrive  later) und you are checking for a value larger then 0.
Can you change it that you also accept >=0 for acenergy ? Then the integration should work also with new installations...

mpredfearn, 2023-05-12:
That would defeat the point of the test, which is to sanity check that the value is legal. For all inverters that have ever had panels attached, that value is greater than 0. Yours is rather a niche case where you're trying to get the integration to work with a panel before it is even commissioned....

ItsMe00007, 2023-05-12:
See above

closed 2023-05-12 by ItsMe00007