binsentsu/home-assistant-solaredge-modbus #165: Battery shows up with absurd voltage, other values don't match with cloud/app data

woomiez, 2023-06-01:
I'm running a pair of SE7K inverters, one hybrid with an 48V BYD BattBox attached (master), the other one PV-only (slave), and the usual Modbus metering device. However, the battery consistently shows up with ~790 Volts. What's wrong there?
Also, values reported to HA don't seem to match with values reported by the SolarEdge cloud portal or app.. e.g. right now the app tells me I'm sending 550W to the house (which matches the equipment I've got running), while HA tells me "AC Power 260W". Could this be an issue with there being two inverters? I would have expected this to be the aggregated value of both.

jovana, 2023-06-01:
If you running multi inverter than is this a better solution for you: 

https://github.com/WillCodeForCats/solaredge-modbus-multi

You give this a try.

ugrolei, 2025-01-01:
Hallo, ich bin neu hier. Das Problem mit dem falschen Spannungswert der Batterie ist mir auch aufgefallen. Wie kann eine 48V Batterie 811V haben? Wo kann ich die aktuelle Solarproduktion (Sonneneinstrahlung) in W finden? Es gibt bei mir 108 Entitäten, aber diesen Wert finde ich nicht.