binsentsu/home-assistant-solaredge-modbus #166: Where to configure

CharlPels, 2023-06-10:
In the doc i read

Power control is disabled by default. It can be enabled in the configuration of this integration by setting power_control to true. When power control is enabled number.solaredge_active_power_limit is available in Home Assistant for reading and writing. The actual value of the active power limit is read together with the other values of the inverter.

where do you enable this on and existing modbus entry, what file to edit and what to add there?

binsentsu, 2023-06-12:
Options flow through UI still needs to be implemented. You can enable it manually on existing configuration by editing file .storage\core.config_entries. Add
`"power_control": true` in the data section of integration solaredge_modbus. After that a restart of HA is necessary to activate it.

klacol, 2023-08-20:
@binsentsu : Thanks, works like a charm
[image]

@CharlPels : Here ist the manual, how to change config entries manually. Hope this helps:
https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/171#issuecomment-1615535278

closed 2024-01-04 by binsentsu

Meatballs1, 2024-03-22:
Reloading the integration seemed to work ok without a full reboot? :D