binsentsu/home-assistant-solaredge-modbus #167: Unable to change the active power limit programmatically

jydoulliez, 2023-06-19:
First of all thanks for this very useful component. I am struggling however to change the active power limit of my inverter.

When I limit the active power of my inverter via the UI (slider of the entity number.solaredge_active_power_limit) the value is taken into account by my inverter. It stays at the selected value until I change it. So in my understanding the inverter is configured correctly.

However, if I try to modify the entity state via the page Developer Tools/States (or programmatically) the value changes briefly then comes back at the value defined in the UI. So I end up with this behaviour where my script is trying to reduce the power but it keeps coming back at 100 % :

[image]

How is it possible to change the value programmatically without fighting the value defined in the UI ? Am I doing something wrong ?

binsentsu, 2023-06-22:
Use an automation which calls service input_number.set_value for the power control entity id

jydoulliez, 2023-06-23:
Thanks for your answer. I tried this service to change the value using the developer tools :

[image]

But it doesn't seem to change it (not even temporarily) and it's the only entity related to power control that I can see.

jydoulliez, 2023-06-23:
Oops by using number.set_value instead of input_number.set_value it seems to work (makes sense). Thanks for pointing me in the right direction !

closed 2023-06-23 by jydoulliez