binsentsu/home-assistant-solaredge-modbus #170: modbus read error after update to v1.9.0

gody01, 2023-06-24:
After update to 1.9.0 integration stops working with modbus read error.
It does not recover until HA restart.

Error reading modbus data
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 292, in recv
    ready = select.select([self.socket], [], [], end - time_)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 104, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 195, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 168, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 323, in _transact
    result = self._recv(response_length, full)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 412, in _recv
    result = self.client.framer.recvPacket(expected_response_length)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/__init__.py", line 53, in recvPacket
    return self.client.recv(size)  # pylint: disable=no-member
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 294, in recv
    return self._handle_abrupt_socket_close(size, data, time.time() - time_)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 347, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(10.94.0.241:1502): Connection unexpectedly closed 2.1219253540039062e-05 seconds into read of unbounded read bytes without response from unit before it closed connection

gody01, 2023-06-26:
I downgraded to V1.8.0 code and it seems to fix the problem.

It looks like mine Solaredge inverter has problem with reading Power Limiter data as with diff between V1.8.0 and V1.9.0 only added function is read_modbus_power_limit.

Sensor seems to be enabled by default and some time after changing to V1.9.0 code simptoms start to happen.

It looks, like Inverter's CPU become overloaded at this read function - as also normal ping communication starts to be very unreliable after above errors in communications is reported.

ItsMe00007, 2023-06-27:
I just startet to use Modbus TCP abou a week ago with version 1.9.
Basicly it runs fine but a few times a day the connection is refused by the inverter.
WLAN connection is stable but i am planning to try an Ethernet Cable just to be sure that IP connection is fine.
For me it looks more that the Inverter has a limit with the number of Bytes transferred during a specific time via Modbus.
How frequently do you poll the inverter ?
Is there place in the script where i can adjust it to check if that is the cause ?

Logger: pymodbus.client.tcp
Source: custom_components/solaredge_modbus/__init__.py:258 
Integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus)) 
First occurred: June 26, 2023 at 14:20:47 (43 occurrences) 
Last logged: 15:44:50

Connection to (192.168.50.140, 1502) failed: [Errno 111] Connection refused
Connection to (192.168.50.140, 1502) failed: timed out
Connection to (192.168.50.140, 1502) failed: [Errno 113] Host is unreachable

gody01, 2023-06-27:
> I just startet to use Modbus TCP abou a week ago with version 1.9. ...
My inverter doesn't even have Modbus on WiFi any more (after some firmware upgrade).

Do You have some other modbus device which connects to inverter ?
Inverter only allows 1 connection to modbus, so if another program has connection it would not answer.

ItsMe00007, 2023-06-27:
> Do You have some other modbus device which connects to inverter ? Inverter only allows 1 connection to modbus, so if another program has connection it would not answer.

No, only the HA connects via Modbus to the Inverter. 
I also have only the Inverter from SE. No other SE device where the inverter might communicate via modbus.

ItsMe00007, 2023-06-28:
I replaced the WIFI connection with a direct cable connection and still got Modbus errors. Just to be sure that WIFI is not the problem:.

Error reading Modbus Data

This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:258 
Integration: SolarEdge Modbus (documentation) 
First occurred: June 28, 2023 at 23:18:53 (1 occurrences) 
Last logged: June 28, 2023 at 23:18:53

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 104, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 195, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 168, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 323, in _transact
    result = self._recv(response_length, full)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 362, in _recv
    read_min = self.client.framer.recvPacket(min_size)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/__init__.py", line 53, in recvPacket
    return self.client.recv(size)  # pylint: disable=no-member
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 297, in recv
    return self._handle_abrupt_socket_close(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 347, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.50.141:1502): Connection unexpectedly closed 4.267692565917969e-05 seconds into read of 8 bytes without response from unit before it closed connection

gody01, 2023-06-29:
Can You try and disable "Active Power Limit" sensor ?
Perhpas with this sensor disabled, integration would't try to read it via modbus and it wouldn't "upset" inverter.

ItsMe00007, 2023-06-29:
> Can You try and disable "Active Power Limit" sensor ?
> Perhpas with this sensor disabled, integration would't try to read it via modbus and it wouldn't "upset" inverter.

I turned "Active Power Limit" off and keep it connected via Ethernet. Waiting for result.

ItsMe00007, 2023-07-01:
Unfortunately there was still one Modbus Error:
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:258 
Integration: SolarEdge Modbus (documentation) 
First occurred: June 30, 2023 at 22:57:00 (1 occurrences) 
Last logged: June 30, 2023 at 22:57:00

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 100, in read_holding_registers
    :param kwargs: (optional) Experimental parameters.
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 192, in execute
    return self.protocol.execute(request)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 159, in execute
    ):
       
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 306, in _transact
    'Changing transaction state from "SENDING" '
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 341, in _recv
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/__init__.py", line 53, in recvPacket
    return self.client.recv(size)  # pylint: disable=no-member
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 314, in recv
    return b"".join(data)
                       ^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 363, in _handle_abrupt_socket_close
    f"<{self.__class__.__name__} at {hex(id(self))} socket={self.socket}, "
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.50.140:1502): Connection unexpectedly closed 5.030632019042969e-05 seconds into read of 8 bytes without response from unit before it closed connection

ItsMe00007, 2023-07-04:
It seems that nearly all  of the Modbus Errors are related to the WLAN connection. 
My connection has an RSSI of -62 which should be enough for a stable connection, but that doesn't seem to be the case.

I now used a direct ethernet cable between the inverter and the Gigabit-Switch where the HA is connected.
I left "Active Power Limit" enabled.

Up to now there was only a single error within 24 hours and that was around 3-oClock in the morning.
I kept a tcpdump running on HA and it seems that the inverter did not respond fast enough.
The time between packet 8335 "Read Holding Registers" and 8336 where HA sends a FIN to end the TCP session is just a few milliseconds.

Then with packet 8346 (I filtered for port 1502) the TCP session is started again.

So as a conclusion: Don't use the WLAN connection of the Inverter.

[image]
[image]

ItsMe00007, 2023-07-22:
Strange thing, I am getting this "Error reading modbus data" describe in my previous poet every day around 11 pm. I guess the inverter is doing some housekeeping at that time and therefore answers to slow at the modbus requests.

gody01, 2023-07-26:
For me upgrade to 1.9.1 seems to resolv the issue.
After upgrade it seems to be working without errors.