binsentsu/home-assistant-solaredge-modbus #171: How to change the IP-Adress of a running integration

klacol, 2023-06-30:
Hello, 
I had to reorganize my local network. The inverter got a new IP-Address. Is it possible to change the address of an existing integration, or do I have to delete an add the interver?

binsentsu, 2023-06-30:
Search for solaredge_modbus in file .storage\core.config_entries. There you will find current ip. Replace it with new ip and reboot HA.

klacol, 2023-07-01:
Thank you. This worked for me on Windows:

- Install the Add-on ``Samba share``
- Start File-Explorer and open ``\\ip-address-of-homeassistant\``
- Enter username and password from the Add-on ``Samba share``
- Enable ``Show hidden items`` in File-Explorer (``View``=>``Show``=>``Hidden items``)
- Edit the (hidden) file ``\\ip-address\config\.storage\core.config_entries`` 
- Search for ``solaredge_modbus`` change the entry ``host`` to the new ip address
- Save the file
- Restart Homeassistant

closed 2023-07-01 by klacol