binsentsu/home-assistant-solaredge-modbus #176: Issues loading integration with current HA Version 2023.7.1 / 2023.7.2

ChristophCaina, 2023-07-14:
Hi,
I am getting the following error while HA is trying to load the integration:

```
Logger: homeassistant.setup
Source: custom_components/solaredge_modbus/__init__.py:12
Integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: 06:58:27 (1 occurrences)
Last logged: 06:58:27

Setup failed for custom integration solaredge_modbus: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 12, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 13, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
```

ItsMe00007, 2023-07-14:
I too have the following problem after migrating HA from 6.x to 7.2:

This error originated from a custom integration.

Logger: homeassistant.setup
Source: custom_components/solaredge_modbus/__init__.py:12 
Integration: solaredge_modbus (documentation) 
First occurred: 10:41:48 (1 occurrences) 
Last logged: 10:41:48

Setup failed for custom integration solaredge_modbus: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 12, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 13, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)

ItsMe00007, 2023-07-14:
It seems that the automatic HA restart after the upgrade was not enough, after rebooting the complete device the Modbus integration works again.

ChristophCaina, 2023-07-14:
hm... yes, completely rebooting the host seems to have solved the issue (including the one I faced with ZHA toolkit)... remaining only one issue with ZHA toolkit, which will be fixed soon.

ChristophCaina, 2023-07-14:
It seems that there is an general issue since one of the 2023.7. releases...

this error has been reported several times now for different components.
- ZHA
- SolarEdge Modbus
- Solax Modbus

binsentsu, 2023-07-17:
HA core pymodbus was upgraded to version 3.3.1 in HA core 2023.7, possible that this has something to do with it. Will check asap.

binsentsu, 2023-07-17:
pymodbus version was aligned with HA core pymodbus version within release V1.9.1. Should be fixed

closed 2023-09-12 by binsentsu