binsentsu/home-assistant-solaredge-modbus #184: Connection Error message not catched

legantois, 2023-08-06:
i regularly get this error sequence (i removed my IP address in the error messages)

i am connected to the Solar Edge usibg a Ethernet connection (not wifi)

Could be a better way to treat this use case😀/error ?

2023-08-05 22:38:36.686 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 104, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 195, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 168, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 323, in _transact
    result = self._recv(response_length, full)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 362, in _recv
    read_min = self.client.framer.recvPacket(min_size)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/__init__.py", line 53, in recvPacket
    return self.client.recv(size)  # pylint: disable=no-member
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 297, in recv
    return self._handle_abrupt_socket_close(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 347, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(): Connection unexpectedly closed 3.4332275390625e-05 seconds into read of 8 bytes without response from unit before it closed connection

legantois, 2023-08-27:
still receiving errors. @binsentsu. How can i help ?

2023-08-26 22:43:31.052 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 274, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 560, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 258, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 102, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 176, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 170, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 320, in _transact
    result = self._recv(response_length, full)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 351, in _recv
    read_min = self.client.framer.recvPacket(min_size)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/base.py", line 68, in recvPacket
    return self.client.recv(size)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 217, in recv
    return self._handle_abrupt_socket_close(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 264, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.178.61:1502): Connection unexpectedly closed 2.86102294921875e-05 seconds into read of 8 bytes without response from slave before it closed connection
2023-08-26 22:44:01.163 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 276, in read_modbus_data
    and self.read_modbus_data_meter1()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 286, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 435, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 148, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

ChristophCaina, 2023-12-04:
these are two different errors!

the first one:

```
Error: [Connection] ModbusTcpClient(): Connection unexpectedly closed 3.4332275390625e-05 seconds into read of 8 bytes without response from unit before it closed connection
```
indicates a lost connection to your inverter - which could be caused by a network issue...

the second one:
```
validate
raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)
```

reports, that a sensor (the data_meter_exported) is reporting a 0 which is unexpected.
The integration is checking for 0 values on sensor that should not report a zero - like the exported kwh sensor.
allowing this zero to be written to the sensor values would cause a major impact on the generated statistics.

This error usually comes up, when the inverter is in (going into) sleep / night mode - according to the time it was logged.
Therefore, I am pretty sure, this error can be ignored and a possible solution would be, to not log this occurance.

ChristophCaina, 2024-01-06:
#206 / #191

legantois, 2024-01-27:
seems the first error is still poping up. Difficult to think it is caused by a network issue as inverter and HA instance are both wired

2024-01-27 05:28:29.089 ERROR (SyncWorker_5) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 247, in _refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 321, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 608, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 305, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 102, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 177, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 168, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 313, in _transact
    result = self._recv(response_length, full)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 343, in _recv
    read_min = self.client.framer.recvPacket(min_size)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/base.py", line 77, in recvPacket
    return self.client.recv(size)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 210, in recv
    return self._handle_abrupt_socket_close(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 257, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.178.61:1502): Connection unexpectedly closed 0.0001239776611328125 seconds into read of 8 bytes without response from slave before it closed connection