binsentsu/home-assistant-solaredge-modbus #186: Export control site limit set too low

chaffis75, 2023-08-13:
When using the the Export control site limit the max value is set too low. 
My sugestion is to increse this value significantly. I have incresed it to 20000 myself since i have a 17K system (some margin). 

The issue is that you can ony put it to 10000 w today, this is too low and will limit the system. (but you can readout larger nymbers already today.)

This is where the change needs to be:
EXPORT_CONTROL_NUMBER_TYPES = [
    ["Export control site limit", "export_control_site_limit", 0xE002, "f", {"min": 0, "max": **10000**, "unit": "W"}],
]

This line is found the the constants file.

binsentsu, 2023-09-16:
Fixed through #198 in [V1.10.0](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.10.0)

closed 2023-09-16 by binsentsu

mmagdalenc, 2025-01-30:
I have the same issue with v2.0.0: "Value 16000 for number.mfe_export_control_site_limit is outside valid range 0 - 10000."

binsentsu, 2025-02-01:
Was idd reintroduced in V2. Now fixed in v2.0.2