binsentsu/home-assistant-solaredge-modbus #190: Vendor error code value not present in SolarEdge's documentation

ljarbinet, 2023-08-22:
Hello,

I'm receiving quote often a 195 "status vendor" value
Unfortunately that value is not describes in the possible vendors codes listed in the installation guide.
Any chance this value would be encoded differently than others ?
Any idea what could cause it ? And where to find it's meaning?

Thanks
Laurent

ljarbinet, 2023-08-22:
Related to these events I notice a production drop at the exact same timings (on HA only, SolarEdge app polling frequency misses them)
[image]

ChristophCaina, 2023-12-04:
The vendor codes can be different depending on the inverter, firmware and so on, therefore, the vendor codes are hard to "translate" ...

Have you already contacted solar edge support?
They should be able to have a look on your inverter and check, what these codes mean.