binsentsu/home-assistant-solaredge-modbus #191: ValueError: Value 0 failed validation (>0)

Maikel-K, 2023-08-28:
Hello,

The intergration has worked perfectly for some time and now is not reading all the data all the time with these errors in the log:

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 216, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 276, in read_modbus_data
    and self.read_modbus_data_meter1()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 286, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 435, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 148, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

Maikel-K, 2023-08-30:
I found another post regarding same error, #114 involved this so i did what that post said. I now have to wait if the error returns.

I still have problems that not all values update everytime, sometimes there are a few hours between updates.
How can i see if there is an issue between my meter and the inverter or the inverter and this intergration?

I also have this error after turning debug logging on.

Entity sensor.solaredgemodbus_m1_imported_kwh from integration solaredge_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 252.982 (254.402) with last_updated set to 2023-08-30T19:50:18.861421+00:00. Please report it to the custom integration author.

ChristophCaina, 2023-12-04:
raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

the integration is trying to validate the "m1 exported" value.
The sensor value should be greater than 0 -> but the inverter is sending a 0 value, therefore, the validation failed.

THe problem here is, that if we would validate for > = 0 - the integration would allow the zero value, and then the integration will fail when the inverter is sending a value greater zero, because it would just add the new value to the previous one, causing issues with the energy dashboard and statistics.

MAYBE a solution to this would be, not to write this into the logs... at least, when this behave is expected.

the second error indicates, that the inverter has sent a smaller value for the imported_kwh sensor, than earlier... which should not happen, based on the fact, that this sensor is continously increasing

ChristophCaina, 2024-08-01:
hm... I can remember that we tried to fix the issue by changing the validation to compare not only if value is > 0 but also for value >= 0 which caused other issues.

I am pretty sure, that the validation itself is not an issue.
the check itself should be OK - and the issue probably only happens, when the integration will be reloaded while the inverter cannot be reached (either connection issue, or night mode [just as in my case right now])...

So this exception should only be logged in some certain conditions, when it could be expected, that the export value is zero.
in Every other case, the used validation is probably correct as far as I can remember from back then.

as already mentioned in my previous comment, it should be considered, if we really need to raise this as an exception in the logs...