binsentsu/home-assistant-solaredge-modbus #192: Meter readings (conected thorugh modbus) halt intermitently while battery is charging (connected through solaredge home network)

voleivol7, 2023-09-06:
I had an installation with an inverter conencted through ethernet and meter conected through modbus to inverter and have been getting readings  and controlling the inverter through modbus tcp without issues since I started to use this integration.
I have now added a battery which is connected to the inverter through solaredge home netwok (some sort of proprietary zigbee as I understand) and are able to get battery readings and control battery modes without issues.
However, now whenever the battery is charging the integration intermitently stops getting readings from the meter (reads ok a couple of cycles, drops a couple of cycles and so on). To diagnose the problem I have traced the chart of meter power AC and I have seen that values goes to 0 intermitently even though am exporting. I have tried different reading cycles (5s, 10s, 30s, 1 min) and the behaviour is the same: one or two reads and no reads for next  couple of cycles. Inverter and the battery do not suffer this problem and get their values on every reading cycle and as soon as the battery stops charging (iddle status, have not diagnosed while discharging yet) the problem dissapears and I get continuous readings from the meter also.
Not sure if anyone with a similar setup (meter through modbus and battery through solaredge home network) is facing a similar problem, but it completely distorts production and consumption information in homeassistant.

ChristophCaina, 2023-12-04:
have you contacted the solar edge support?
For me, this sounds like an issue on their end... if the inverter does not provide such information over modbus during charging.

Also, please check if your logs contain any additional information that might help :)
This could at least help to identify if this is an issue within the integration (which I don't think it will be)

voleivol7, 2024-01-01:
It was ultimately a problem in my HA setup not related to your integration.
Thanks and apologies.