binsentsu/home-assistant-solaredge-modbus #194: pymodbus Endian.Little not found issue on 2023.9.1

mp3guy, 2023-09-09:
On import of this custom component, somehow it fails to find this Enum (python3.11, if that makes a difference) when importing pymodbus types. Had to downgrade to 2023.7.0.

clsgroen, 2023-09-09:
Same issue. Also 2023.9.1

SpeedlimitsB, 2023-09-09:
Same here ....

This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/solaredge_modbus/__init__.py:12
Integration: solaredge_modbus (documentation) 
First occurred: 23:15:45 (1 occurrences) 
Last logged: 23:15:45

Unexpected exception importing component custom_components.solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 12, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

SpeedlimitsB, 2023-09-10:
Just did a complete reboot ( power off/on) and it works now . 

Home Assistant 2023.9.1
Supervisor 2023.08.3
Operating System 10.5
Frontend 20230908.0 - latest

lka, 2023-09-10:
Home Assistant 2023.9.1
Supervisor 2023.08.3
Operating System 10.5
Frontend 20230908.0 - latest

2023-09-09 18:00:56.262 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: Big
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 293, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 194, in _async_update_data
return await self._refresh_modbus_data_with_retry(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 232, in _refresh_modbus_data_with_retry
raise ex
File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 229, in _refresh_modbus_data_with_retry
return await self._hub.async_refresh_modbus_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 384, in async_refresh_modbus_data
await self._async_init_solaredge()
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 202, in _async_init_solaredge
await new_inverter.init_device()
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 727, in init_device
inverter_data.registers, byteorder=Endian.Big
^^^^^^^^^^
File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
raise AttributeError(name) from None
AttributeError: Big

Complete restart of the System didn‘t help.

lka, 2023-09-10:
Just updated to Version 2.4.5 and it works again 
thank you very much

mp3guy, 2023-09-10:
Updated what to 2.4.5? I'm running HA Core, and a full reboot didn't work.

lka, 2023-09-10:
I was Running solaredge_modbus_multi with version 2.4.4 having the above stated issue. The update to version 2.4.5 did the work. The byteorder=Endian.Big caused the error. In the new version I found byteorder=Endian.BIG and that is correct.

mxwi, 2023-09-11:
you are talking about solaredge_modbus_multi. That is another addon. This one needs to be updated, too.

NdS-Research-Facilities, 2023-09-11:
Same issue:

 _Home Assistant 2023.9.1 Supervisor 2023.08.3 Operating System 10.5 Frontend 20230608.0 - latest_ 

```
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/solaredge_modbus/__init__.py:12
Integration: solaredge_modbus (documentation)
First occurred: September 10, 2023 at 19:27:12 (1 occurrences)
Last logged: September 10, 2023 at 19:27:12

Unexpected exception importing component custom_components.solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 12, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

```

binsentsu, 2023-09-12:
Fixed in release [V1.9.2](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.9.2)

closed 2023-09-12 by binsentsu