binsentsu/home-assistant-solaredge-modbus #196: Abrupt changes in Imported or Exported Metering kWh

yannik1015, 2023-09-11:
Hey every now and then the inverter seems to adjust the values for total imported and exported kWh I don't know if this is some sort of "self-correction" with some outside meter (Maybe the one used for billing by the electricity company?!?) or some strange quirk in the inverter itself.

Here you can see an example of the energy dashboard showing clearly something went wrong between 21:00-22:00
[image]

The following two pictures show both sensor values on their own.
[image]
[image]

These jumps are sometimes "small" around 7 kWh but can also be quite large like in this case. When looking at the Solaredge monitoring site these don't show up.
[image]

### Possible solution
As these changes seem to be permanent I think one option for fixing this would be to now allow these total values to decrease and if they do the difference should be saved to an offset value, that gets added from their on now to the sensor value, to achieve the correct value.

Issue most likely linked to #174

Integration-version 1.9.0

tmp-hallenser, 2023-10-18:
Just had the same behavior

ChristophCaina, 2023-12-04:
Are you also using the Core SolarEdge integration?
If yes, the "Inverter" Sensor does provide some information about the Firmware version running...

This integration probably just use the values, the inverter is providing, and if this happens more randomly, I think there's not much the integration could do to prevent that from happening.

I don't know if such an issue can happen, if the meter would be "reset" or "restarted" somehow... or if it is loosing the connection to the inverter and then, restoring it again.

Personally, I haven't recognized such cases - the only case I do have right now is a big "increase" for this sensor - caused by a Crash of my Database - and the integration of "longterm statistics" into the history card.

[image]

yannik1015, 2023-12-04:
It just seems to affect the total values from the inverter. 
My solution was to create an integration sensor with method left on the current imported and exported power and created two utility meter entities using these. That does work without a problem.