binsentsu/home-assistant-solaredge-modbus #201: All entities become 'Unknown' after upgrade to v1.10.0

theautomation, 2023-09-22:
All my solaredge modbus entities become 'Unknown' after upgrade to v1.10.0 any idea what this could be? 

```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:227
Integration: SolarEdge Modbus (documentation)
First occurred: 7:34:58 AM (1653 occurrences)
Last logged: 8:02:32 AM

not able to connect to 10.0.30.90:1502
```

AD-Tweak, 2023-09-22:
the same here
After the update v1.10.0, I can no longer connect to my 3-phase inverter
This problem does not occur with my 1-phase inverters and everything functions fine

joenivl, 2023-09-22:
Same here (1 phase converter (SE4000H HD-Wave) with modbus over wifi). Downgraded to 1.9 and it works again.

binsentsu, 2023-09-22:
Can you share the polling interval which is configured for the integration?

theautomation, 2023-09-22:
> Can you share the polling interval which is configured for the integration?

I have set that to 1 sec

AD-Tweak, 2023-09-22:
set it to 1second

joenivl, 2023-09-23:
1 sec

binsentsu, 2023-09-23:
That will be the reason. With PR #183, the connect timeout for the modbus connection is now correlated to the scan_interval. But with a scan_interval if 1 the connect_timeout basically becomes 0, which is not enough time to complete the connection. Will make a patch for it.

binsentsu, 2023-09-25:
Should be fixed in [V1.10.1](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.10.1)

theautomation, 2023-09-25:
> Should be fixed in [V1.10.1](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.10.1)

Yes, it is working again! Thanks for this intergration!

closed 2023-09-25 by theautomation

joenivl, 2023-09-26:
> Should be fixed in [V1.10.1](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.10.1)

Works again, thanks voor fixing!!

AndreaDellaRosa, 2023-11-08:
How do i fix? step by step please

theautomation, 2023-11-08:
> How do i fix? step by step please

Use [V1.10.1](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.10.1)