binsentsu/home-assistant-solaredge-modbus #204: v1.10.x is intefering with other integrations

ExcelsiorEyer, 2023-09-27:
Upgrading from 1.9.1 causes several other core integrations to fail. If I disable solaredge modbus device or downgrade it to 1.9.1, the issue goes away. As you can see, the logs there is several time-outs observed but nothing points to an obvious error in the logs.

```
2023-09-27 17:59:32.620 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.zone_14 is taking over 10 seconds
2023-09-27 17:59:32.632 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.br_tv is taking over 10 seconds
2023-09-27 17:59:32.634 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching KasaPowerStrip.home.arpa data: Unable to connect to the device: KasaPowerStrip.home.arpa:9999: 
2023-09-27 17:59:32.649 WARNING (MainThread) [homeassistant.components.media_player] Updating volumio media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:32.650 WARNING (MainThread) [homeassistant.components.media_player] Updating monoprice media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:32.652 WARNING (MainThread) [homeassistant.components.media_player] Updating webostv media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:32.659 WARNING (MainThread) [homeassistant.components.media_player] Updating alexa_media media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:32.661 INFO (MainThread) [custom_components.solaredge_modbus] modbus client is not connected, trying to reconnect
2023-09-27 17:59:32.663 INFO (MainThread) [custom_components.solaredge_modbus] successfully connected to 10.20.13.22:502
2023-09-27 17:59:50.668 DEBUG (MainThread) [custom_components.solaredge_modbus] Could not read Active Power Limit
2023-09-27 17:59:50.707 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.zone_15 is taking over 10 seconds
2023-09-27 17:59:50.717 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.22 data: Unable to connect to the device: 10.20.12.22:9999: 
2023-09-27 17:59:50.719 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.21 data: Unable to connect to the device: 10.20.12.21:9999: 
2023-09-27 17:59:50.720 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.27 data: Unable to connect to the device: 10.20.12.27:9999: 
2023-09-27 17:59:50.721 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.28 data: Unable to connect to the device: 10.20.12.28:9999: 
2023-09-27 17:59:50.723 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.24 data: Unable to connect to the device: 10.20.12.24:9999: 
2023-09-27 17:59:50.724 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.26 data: Unable to connect to the device: 10.20.12.26:9999: 
2023-09-27 17:59:50.725 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.25 data: Unable to connect to the device: 10.20.12.25:9999: 
2023-09-27 17:59:50.726 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.23 data: Unable to connect to the device: 10.20.12.23:9999: 
2023-09-27 17:59:50.730 WARNING (MainThread) [homeassistant.components.media_player] Updating volumio media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:50.732 WARNING (MainThread) [homeassistant.components.media_player] Updating monoprice media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:50.733 WARNING (MainThread) [homeassistant.components.media_player] Updating webostv media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:50.735 WARNING (MainThread) [homeassistant.components.media_player] Updating alexa_media media_player took longer than the scheduled update interval 0:00:10
2023-09-27 17:59:50.736 INFO (MainThread) [custom_components.solaredge_modbus] modbus client is not connected, trying to reconnect
2023-09-27 17:59:50.739 INFO (MainThread) [custom_components.solaredge_modbus] successfully connected to 10.20.13.22:502
2023-09-27 18:00:08.751 DEBUG (MainThread) [custom_components.solaredge_modbus] Could not read Active Power Limit

```

The log entry for _Could not read Active Power Limit_ has not been an issue because of my inverter configuration and appears in earlier version as well.

All components on the latest stable version.

Hammertime84, 2023-10-06:
I have exactly the same problem! Downgrading to version 1.9.2 seems to resolve the timeout issues.

ExcelsiorEyer, 2023-12-29:
> > Upgrading from 1.9.1 causes several other core integrations to fail. If I disable solaredge modbus device or downgrade it to 1.9.2, the issue goes away. As you can see, the logs there is several time-outs observed but nothing points to an obvious error in the logs.
> > ```
> > 2023-09-27 17:59:32.620 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.zone_14 is taking over 10 seconds
> > 2023-09-27 17:59:32.632 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.br_tv is taking over 10 seconds
> > 2023-09-27 17:59:32.634 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching KasaPowerStrip.home.arpa data: Unable to connect to the device: KasaPowerStrip.home.arpa:9999: 
> > 2023-09-27 17:59:32.649 WARNING (MainThread) [homeassistant.components.media_player] Updating volumio media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:32.650 WARNING (MainThread) [homeassistant.components.media_player] Updating monoprice media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:32.652 WARNING (MainThread) [homeassistant.components.media_player] Updating webostv media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:32.659 WARNING (MainThread) [homeassistant.components.media_player] Updating alexa_media media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:32.661 INFO (MainThread) [custom_components.solaredge_modbus] modbus client is not connected, trying to reconnect
> > 2023-09-27 17:59:32.663 INFO (MainThread) [custom_components.solaredge_modbus] successfully connected to 10.20.13.22:502
> > 2023-09-27 17:59:50.668 DEBUG (MainThread) [custom_components.solaredge_modbus] Could not read Active Power Limit
> > 2023-09-27 17:59:50.707 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.zone_15 is taking over 10 seconds
> > 2023-09-27 17:59:50.717 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.22 data: Unable to connect to the device: 10.20.12.22:9999: 
> > 2023-09-27 17:59:50.719 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.21 data: Unable to connect to the device: 10.20.12.21:9999: 
> > 2023-09-27 17:59:50.720 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.27 data: Unable to connect to the device: 10.20.12.27:9999: 
> > 2023-09-27 17:59:50.721 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.28 data: Unable to connect to the device: 10.20.12.28:9999: 
> > 2023-09-27 17:59:50.723 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.24 data: Unable to connect to the device: 10.20.12.24:9999: 
> > 2023-09-27 17:59:50.724 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.26 data: Unable to connect to the device: 10.20.12.26:9999: 
> > 2023-09-27 17:59:50.725 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.25 data: Unable to connect to the device: 10.20.12.25:9999: 
> > 2023-09-27 17:59:50.726 ERROR (MainThread) [homeassistant.components.tplink.coordinator] Error fetching 10.20.12.23 data: Unable to connect to the device: 10.20.12.23:9999: 
> > 2023-09-27 17:59:50.730 WARNING (MainThread) [homeassistant.components.media_player] Updating volumio media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:50.732 WARNING (MainThread) [homeassistant.components.media_player] Updating monoprice media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:50.733 WARNING (MainThread) [homeassistant.components.media_player] Updating webostv media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:50.735 WARNING (MainThread) [homeassistant.components.media_player] Updating alexa_media media_player took longer than the scheduled update interval 0:00:10
> > 2023-09-27 17:59:50.736 INFO (MainThread) [custom_components.solaredge_modbus] modbus client is not connected, trying to reconnect
> > 2023-09-27 17:59:50.739 INFO (MainThread) [custom_components.solaredge_modbus] successfully connected to 10.20.13.22:502
> > 2023-09-27 18:00:08.751 DEBUG (MainThread) [custom_components.solaredge_modbus] Could not read Active Power Limit
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > The log entry for _Could not read Active Power Limit_ has not been an issue because of my inverter configuration and appears in earlier version as well.
> > All components on the latest stable version.
> 
> Does this also cause frontend issues? I just downloaded this and noticed the frontend plus integrations blow up and become very slow. Once I disabled it, everything was fine. Something is blocking i/o.

Yes it does, and that's the reason why I'm still not upgrading. You could try downgrading it to the last 1.9.x version and you should be fine.

binsentsu, 2023-12-31:
Only difference in this regard between 1.10 and 1.9 is that the connect timeout is now based on the scan interval while it used to be 3 seconds (the default of the modbus client implementation). What's your configured update interval for the sensor readings?

lits101, 2024-01-01:
I was having this issue with 1.10.x as well, to the point where things like zigbee motion sensors weren't updating their status, automations weren't firing, lights weren't turning on... 

I've downgraded to 1.9.2 and hoping that fixes it.

ExcelsiorEyer, 2024-01-03:
> Please check if V1.10.2 fixes this issue for you

Upgraded to v1.10.2 and I'm not seeing the issue anymore.