binsentsu/home-assistant-solaredge-modbus #205: Entities not available to add to items on energy dashboard

dsmck58, 2023-10-01:
The integration shows that all the sensors have been added and at the moment they appear in a list on my Overview page..  However, when I try to add specific sensors to something in the energy dashboard, only 1 sensor AC Power, is available.  None of the other sensors appear. New user so maybe simple solution. thanks

loryanstrant, 2023-10-07:
Same here. Just got it running today but don't see them as sensors to be added into the energy dashboard.
I created the templated sensors from this page: https://github.com/binsentsu/home-assistant-solaredge-modbus/wiki/Using-Templated-Sensors-to-Calculate-Power-Flow-and-Energy#single-inverter-single-meter-at-grid-import--export-point

deruesch, 2023-10-08:
Same here. I cant create a functional dashboard w/o the sensors. especially the sensor for the production is not available. 
Is there a way to fix it?
Thx.

deruesch, 2023-11-21:
Hello, is there a Chance that we get an Update? 
It's unusable.

dsmck58, 2023-11-21:
Not sure what you want me to update.  If I had the answer, I would not have raised the question.

mpredfearn, 2023-11-21:
You need to create the correct set of templated sensors for your particular system set up (see above link to wiki). The raw sensors that the inverter / meters provide don't give you all of the necessary sensors for the energy dashboard. The integration does not / cannot do this for you because it can't automatically figure out how your particular system has been wired and configured.

dsmck58, 2023-11-21:
Thanks Matt. Clearest explanation I have seen! Now I just have to figure it all out but I see where you are coming from.