binsentsu/home-assistant-solaredge-modbus #206: Modbus creating system erros

w1Ngx, 2023-10-05:
Hi,

I am using modbus since a week and noticed there are couple of errors produced:. Is this something i can fix myself?

Thanks a lot!

1:
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:232
Integration: SolarEdge Modbus (documentation)
First occurred: 7:00:07 AM (3 occurrences)
Last logged: 7:00:16 AM

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 232, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 309, in read_modbus_data
    and self.read_modbus_data_meter1()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 319, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 468, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 158, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

2:
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:232
Integration: SolarEdge Modbus (documentation)
First occurred: 7:00:05 AM (1 occurrences)
Last logged: 7:00:05 AM

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 232, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 307, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 674, in read_modbus_data_inverter
    self.data["acenergy"] = round(acenergy * 0.001, 3)
                                  ~~~~~~~~~^~~~~~~
OverflowError: int too large to convert to float

branda92, 2023-10-30:
Hi, I also have the same problem, I don't know if it's been a long time or the latest versions. If anyone can help us fix the errors it would help us a lot.

duracellkonijn22, 2023-11-22:
Same here... any update about this?

ChristophCaina, 2023-12-04:
it seems, that the inverter is reporting "0" for the sensor meter_exported sensor, but the code is trying to validate the value, because it is checking for >0 

I remember, that we tried to fix this - or something similar in the past, but this caused other issues when the inverter went to sleep mode - and woke up on the next day.

The second error seems to be an issue with the data type - I am not sure, if the inverter is sending an integer value, and the integration is trying to convert it to a float value...

I need to check, if I do have the same errors in my logs...

ChristophCaina, 2023-12-04:
I can verify that I also do have the issue about the validation check for Meter1_exported:

```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:232
Integration: SolarEdge Modbus (documentation)
First occurred: 18:06:05 (1 occurrences)
Last logged: 18:06:05

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 232, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 309, in read_modbus_data
    and self.read_modbus_data_meter1()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 319, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 468, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 158, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)
```

but I do not have the error about the integer to float convert... which could also be due to the value of the sensor...

binsentsu, 2023-12-31:
It seems that sometimes inadvertently the value reported through modbus becomes 0. These validations are now raising exceptions. Mayb it's an option to skip the invalid value and keep the previous value in this case?

ChristophCaina, 2024-01-06:
also related: #191

w1Ngx, 2024-04-12:
Is someone working on the issue? still the same after 6 months

w1Ngx, 2024-07-20:
Is there any was to solve those errors myself? I assume that i am not the only person with those errors, what do others do?