binsentsu/home-assistant-solaredge-modbus #209: Control "Storage Default Mode" Through Automation

dsmck58, 2023-10-10:
Apologies if this is the wrong way to do this but I have a follow on question. I have managed to get to a similar stage setting up my automation.  However, whenever I select a different battery control mode e.g. Charge from PV, it works for a very short period (seconds to minutes) and then automatically switches back to "Maximise self consumption". I have tried everything I can think of but no luck.  

the same thing happens when I select the charge mode "manually" from the Select grouping on the Overview tab . 

Getting very frustrated so would welcome any help.

Thanks

### Discussed in https://github.com/binsentsu/home-assistant-solaredge-modbus/discussions/199

<div type='discussions-op-text'>

<sup>Originally posted by **Darkclad117** September 17, 2023</sup>
Hi all,

I've been using this project for some time and now would like to start controlling my invertor. I've completed the setup and can read all the data and change the settings from the UI, however I'm stuck when trying to do this from an automation. 

How can I set the "select.solaredge_storage_default_mode" entity from an automation? I've tried using Call Service from the automation however I can't see the entity in the Automation UI. Do you have any advise please?

[image]
[image]

Thank you to everyone who has contributed to this project, I'm very grateful!</div>

bradrushworth, 2023-10-24:
This has been happening to me for the last week also. I've tried everything I can think of, but it just keeps happening.

dsmck58, 2023-10-24:
Hey Brad.  I was the same and Solaredge never responded to the query I raised. I just kept trying to override it and then it just stopped happening. No idea what I did I'm afraid so I am no use to you whatsoever. I wish I could be more helpful.

bradrushworth, 2023-11-08:
It seems that:

Storage Default Mode needs to be set to "Maximize self consumption" or whatever you need, first.
Storage Control Mode needs to be set to "Remote Control" second.
Storage Remote Command Mode finally needs to be set to "Charge from PV first" or whatever you want, last.

I had these in a single Scene and this caused it to break. If I don't set the Storage Default Mode and Storage Control Mode in the scene settings, and just set the Storage Remote Command Mode only, then it seems to work.

Finally, note that Storage Remote Command Timeout is usually set to 3600s, which means after an hour the above command will automatically reset from the current Storage Remote Command Mode to whatever is in the Storage Default Mode. This is often desirable behaviour, so I think this is the right way to tackle this issue i.e. if Home Assistant crashes on you, the battery goes back to something sensible in a hour.