binsentsu/home-assistant-solaredge-modbus #210: Multiple drops of communication

Omega00123, 2023-10-28:
Hello I am using the Solar edge integration now for 4 weeks and found it to be a good integration. It dos however create multiple errors in my log and I have not got proof but it seems to make the total system unstable. 172.116.0.114 is my Soloar edge controller.

Here is a copy of part of the log which I see regularly:

File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 262, in _handle_abrupt_socket_close
raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(172.116.0.114:1502): Connection unexpectedly closed 0.00030922889709472656 seconds into read of 8 bytes without response from slave before it closed connection
2023-10-27 20:20:05.767 ERROR (MainThread) [pymodbus.logging] Connection to (172.116.0.114, 1502) failed: [Errno 113] Host is unreachable
2023-10-27 20:20:05.768 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 172.116.0.114:1502
2023-10-27 20:21:01.766 ERROR (MainThread) [pymodbus.logging] Connection to (172.116.0.114, 1502) failed: timed out
2023-10-27 20:21:01.769 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 172.116.0.114:1502
2023-10-27 20:21:18.219 WARNING (MainThread) [haffmpeg.core] Timeout while waiting of FFmpeg
2023-10-27 20:32:29.353 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [140293215605440] from 172.116.0.102 (Home Assistant/2023.7 (io.robbie.HomeAssistant; build:2023.471; iPadOS 15.8.0)): Disconnected: Did not receive auth message within 10 seconds
2023-10-27 21:59:04.269 WARNING (MainThread) [homeassistant.components.sonos.speaker] No recent activity and cannot reach Woonkamer, marking unavailable
2023-10-27 21:59:45.612 WARNING (Thread-5) [pychromecast.socket_client] [Huiskamer(172.116.0.11):8009] Heartbeat timeout, resetting connection
2023-10-27 22:00:15.624 ERROR (Thread-5) [pychromecast.socket_client] [Huiskamer(172.116.0.11):8009] Failed to connect to service ServiceInfo(type='host', data=('172.116.0.11', 8009)), retrying in 5.0s
2023-10-28 06:20:13.652 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
File "/config/custom_components/solaredge_modbus/__init__.py", line 232, in async_refresh_modbus_data
update_result = self.read_modbus_data()

What can it be?

adibis, 2023-11-03:
When you say entire system unresponsive, do you mean home assistant becomes unresponsive and won't let you interact with it?

If yes then that's exactly what I have noticed too. I just got the inverter, installed the integration and my system started to lag/fail to connect/etc

Removing the integration gets rid of this.

I unfortunately don't have logs right now. I'll try to post something tomorrow if I can get it.

ChristophCaina, 2023-12-04:
the logs from the topic opener seem to be a network issue:

```
File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 262, in _handle_abrupt_socket_close
raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(172.116.0.114:1502): Connection unexpectedly closed 0.00030922889709472656 seconds into read of 8 bytes without response from slave before it closed connection
2023-10-27 20:20:05.767 ERROR (MainThread) [pymodbus.logging] Connection to (172.116.0.114, 1502) failed: [Errno 113] Host is unreachable
2023-10-27 20:20:05.768 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 172.116.0.114:1502
2023-10-27 20:21:01.766 ERROR (MainThread) [pymodbus.logging] Connection to (172.116.0.114, 1502) failed: timed out
2023-10-27 20:21:01.769 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 172.116.0.114:1502
2023-10-27 20:21:18.219 WARNING (MainThread) [haffmpeg.core] Timeout while waiting of FFmpeg
2023-10-27 20:32:29.353 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [140293215605440] from 172.116.0.102 (Home Assistant/2023.7 (io.robbie.HomeAssistant; build:2023.471; iPadOS 15.8.0)): Disconnected: Did not receive auth message within 10 seconds
2023-10-27 21:59:04.269 WARNING (MainThread) [homeassistant.components.sonos.speaker] No recent activity and cannot reach Woonkamer, marking unavailable
2023-10-27 21:59:45.612 WARNING (Thread-5) [pychromecast.socket_client] [Huiskamer(172.116.0.11):8009] Heartbeat timeout, resetting connection
2023-10-27 22:00:15.624 ERROR (Thread-5) [pychromecast.socket_client] [Huiskamer(172.116.0.11):8009] Failed to connect to service ServiceInfo(type='host', data=('172.116.0.11', 8009)), retrying in 5.0s
2023-10-28 06:20:13.652 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
File "/config/custom_components/solaredge_modbus/init.py", line 232, in async_refresh_modbus_data
update_result = self.read_modbus_data()
```

This is the Error logged for Modbus:
**Connection unexpectedly closed** 
2023-10-27 20:20:05.767 ERROR (MainThread) [pymodbus.logging] Connection to (172.116.0.114, 1502) failed: [Errno 113] Host is unreachable

But there are also similar issues for other integrations:
```
from 172.116.0.102 (Home Assistant/2023.7 (io.robbie.HomeAssistant; build:2023.471; iPadOS 15.8.0)): Disconnected: Did not receive auth message within 10 seconds
2023-10-27 21:59:04.269 WARNING (MainThread) [homeassistant.components.sonos.speaker] No recent activity and cannot reach Woonkamer, marking unavailable
2023-10-27 21:59:45.612 WARNING (Thread-5) [pychromecast.socket_client] [Huiskamer(172.116.0.11):8009] Heartbeat timeout, resetting connection
2023-10-27 22:00:15.624 ERROR (Thread-5) [pychromecast.socket_client] [Huiskamer(172.116.0.11):8009] Failed to connect to service ServiceInfo(type='host', data=('172.116.0.11', 8009)), retrying in 5.0s
```

There is also the question:
1. Is the inverter connected per ethernet (wired) or wifi?
2. What Firmware versions are running?

--> You can get the information when using the Core-Integration (connected to the Web API) - they are provided as attributes within the "Inverters" Sensor:

[image]

That's pretty sure either a network issue, or something with the inverter (Firmware) - but not a general issue with the integration itself.

mauriziosacca, 2023-12-04:
This is my inverter:
- name: Inverter 1
  manufacturer: SolarEdge
  model: SE3000H-RW000NWN2
  communicationMethod: WIFI
  cpuVersion: 3.2221.0
  SN: *******
  connectedOptimizers: 9

mauriziosacca, 2023-12-04:
Unfortunately I can not. Inverter and router are separated by a wall, without doors or windows, and I have no way to pass an ethernet cable, except through the pipes where the electrical cables pass, and in that case the operation would be complex

TroLoos, 2023-12-24:
Yes, I confirm - there is a serious issue with these hiccups. Whole HA Core becomes unavailable for 30 seconds. It makes whole HA installation very unstable... I started to monitor this unstability with Uptime Kuma, and sensor loads HA webpage. Here's how it looks like on a fresh HA installation (Docker container) with only SolarEdge Modbus integration installed (nothing else at all):

[image]

And here's as a contrary example fully loaded HA instance with no SolarEdge Modbus integration:

[image]

I've spent last couple of that trying to filter out which integration causes such instability... Of course ping to HA host works all the time, only 'homeassistant' container is blocker and freezes.

binsentsu, 2023-12-31:
I think the reason is that the sensor values are currently updated within the main HA thread. As long as there is a quick and stable connection towards the inverter this does not pose a problem. But if connecting/getting the values takes a considerable amount of time, the HA main thread is occupied long enough so it becomes visible through for example unresponsiveness UI. I'll check if the updating can be offloaded to a HA executor job

henrikpieter, 2024-01-01:
Upgrading from 2023.8.4. to 2023.12.3 makes HA slow in responding. Same messages in log as descibed by Mauriziosacca. Doubt if it has anything to do with the network, given that all other integrations work as designed and I don't see any wierd network behaviour.

Re-installing the plug-in doesn't solve the problem. The moment you disable the plug-in, the issues dissapear.

Making the interval period longer gives some releave but problems don't dissapear.

binsentsu, 2024-01-03:
Could you try version V1.10.2 if it's any better?

Meatballs1, 2024-01-04:
The change probably properly fixes https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/100

TroLoos, 2024-01-04:
> The change probably properly fixes #100

Yes, 100% sure it should - what I'm measuring exactly is UI responsiveness (this sensor loads HA login webpage, so it comes from main 'homeassistant' container).

State for now, the fix relieved UI in 100%. I've updated this integration around 5 am, and until now 24h average response time is improving dramatically toward positive side (was 32ms, is already 21ms, it should be completely flat tomorrow morning):

[image]

Omega00123, 2024-01-04:
There is a clear improvement. I have the update running now for 7 Hours and only have 5 missed communications where I had 35 before so 7* better. It is still not zero let’s see how it evolves.

Oscar H.M. De Vries
***@***.***

TroLoos, 2024-01-05:
Yes, definitely now it works as expected:

[image]

Avg. Response for 24-hour time period went down from initial 142ms (2 weeks ago) to 7ms (full day after upgrade to newest version).

Since working thread is now a separate from core HA, perhaps it would be a good idea to give SolarEdge a little bit more time to answer the call? This integration still generates quite a lot of warning/error related to connectivity issues. I think it is not really a matter of device availability on WiFi network (ping works 100% of the time) but rather exclusive access to TCP port 1502. Even if I setup to poll SolarEdge every 30s (or every 5s) - it doesn't make a difference, errors and warnings still appear in a pretty much same amount. I guess it is a matter of handling TCP connection.

binsentsu, 2024-01-05:
@TroLoos Can you share latest logging with connection errors?

TroLoos, 2024-01-07:
@binsentsu actually, turns out that my WiFi link to inverter isn't that solid as I thought... and I guess this is a problem showing up in my logs. I will change the antenna to something better and see if it help. For now - I think this issue has been successfully resolved.

Thank you for your support!