binsentsu/home-assistant-solaredge-modbus #213: [Errno 111] Connection refused

philippegabert, 2023-11-24:
Hi, 

I have been using the integration for quite some time and it works but I always have had some ERROR in the logs. My refresh rate is 1sec. I have the latest version installed (1.10.1).

`2023-11-24 15:22:21.406 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.4.67, 1502) failed: timed out
2023-11-24 15:22:21.407 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 192.168.4.67:1502
2023-11-24 15:22:24.437 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.4.67, 1502) failed: timed out
2023-11-24 15:22:24.437 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 192.168.4.67:1502
2023-11-24 15:22:26.297 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.4.67, 1502) failed: [Errno 111] Connection refused
2023-11-24 15:22:26.298 WARNING (MainThread) [custom_components.solaredge_modbus] not able to connect to 192.168.4.67:1502
2023-11-24 15:22:26.334 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.4.67, 1502) failed: [Errno 111] Connection refused`

Is there anyway to get rid of those errors? As mentioned, the integration works and I do get the data... so I don't know why the integration keeps on throwing those errors.

Thanks

philippegabert, 2023-12-13:
Thanks @ChristophCaina. Question though: It looks to me that it's normal that the inverter refuses the connection as you can only have active.
But... why does the integration tries to connect to the inverter? I assume (but may be wrong) that the "Connection refused" is for "new" connections to the inverter... In the case you have short update frequencies (1 second) and short timeout, the integration should be able to reuse existing connections and therefore not throw those kind of errors?

legantois, 2024-03-17:
#206, #184, #191

Still getting same issue regularly; there should be a way to handle this

2024-03-17 06:43:38.493 ERROR (SyncWorker_31) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 247, in _refresh_modbus_data
    update_result = self.read_modbus_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 321, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 608, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 305, in read_holding_registers
    return self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 102, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 177, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transaction.py", line 168, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transaction.py", line 313, in _transact
    result = self._recv(response_length, full)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transaction.py", line 343, in _recv
    read_min = self.client.framer.recvPacket(min_size)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/base.py", line 77, in recvPacket
    return self.client.recv(size)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/tcp.py", line 210, in recv
    return self._handle_abrupt_socket_close(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/tcp.py", line 257, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.178.61:1502): Connection unexpectedly closed 2.1696090698242188e-05 seconds into read of 8 bytes without response from slave before it closed connection
2024-03-17 06:44:10.546 ERROR (SyncWorker_51) [pymodbus.logging] Connection to (192.168.178.61, 1502) failed: [Errno 111] Connection refused
2024-03-17 06:44:10.547 WARNING (SyncWorker_51) [custom_components.solaredge_modbus] not able to connect to 192.168.178.61:1502

jwindley, 2024-05-13:
I have the same issue. Any update on this please?

willembargeman, 2024-06-18:
Experienced the same kind of issue. Noticed that HA was not able to reach the SolarEdge inverter. In the log files I see the following.

ERROR (MainThread) [pymodbus.client.tcp] Connection to (10.27.11.9, 1502) failed: timed out

Created packet capture on the network and noticed that the inverter was not responding to the TCP SYN request. After some additional testing I found a interesting behaviour.

The inverter was drop all incoming requests from the same subnet. In my example, the inverter and HA are both placed in the subnet 10.27.11.0/24. (so 10.27.11.0-10.27.11.255). Even ping traffic is dropped. 

With my laptop I was able to ping the inverter and also got a SYN ACK when using NMAP on port 1502. My laptop was in a different segment (10.27.10.0/24). 

Workaround is by placing HA and the inverter in a different subnet or by using NAT (which is more complex).

Doc-Jump, 2024-11-09:
Same here, Integration suddenly stopped working, aswell SolarEdge Modbus and also SolarEdge Modbus Multi integration.
Tried to disable and enable the TCP Modbus setting in the inverter's interface, doesn't help. 
I also think it must have to do something with the converter instead of the integration as the inverter can be reached over network but refuses connection on the Modbus-Port. Did anyone contact SolarEdge's support?

Doc-Jump, 2024-11-13:
Deal all,

Had contact with Solaredge support. 
They remotely installed a newer firmware on the inverter. Since then it has been running smoothly again. Simply report it there with the reference to error code 111 and that no data can be accessed via the Modbus. 

Übersetzt mit DeepL (https://www.deepl.com/app/?utm_source=ios&utm_medium=app&utm_campaign=share-translation)

ChristophCaina, 2025-01-28:
> In my opinion, this issue could be closed, as it seems to be a Software Problem on the inverter.

which is implied by the error text `connection refused`
A `connection refused` message always happens, if the remote device you want to connect is activly refusing the attempt to establish the connection.

Nothing, that can be done within the integration.

closed 2025-01-28 by binsentsu

K-Raymer, 2025-01-28:
I was able to resolve the issue using a different local integration, with no software update to the inverter. This to me implies it could be an issue the integration.