binsentsu/home-assistant-solaredge-modbus #232: Power control gets overruled

RedWheiler80, 2024-04-29:
Thank you for your work.

I've connected my inverter with home assistant and everything works great except...

When I change the number for the 'power control' it does work, but gets overruled by the inverter itself after a couple of seconds back to 100%.

Seems like I'm overlooking some setting I think...

Stievo9997, 2024-04-30:
I've got exact same issue. It worked before, but now for some reasons it's get overruled after a few seconds and is goes back to 100%

knilessew, 2024-05-01:
Same over here. Even if I set the value on the inverter itself.. HA keeps it at 100%.

gijsrrr, 2024-05-01:
Same over here, nothing in the debug logs from HA that it's "resetting" to 100.

sk72, 2024-05-01:
Isn't this related to this remark in the documentation?
Power control is disabled by default. It can be enabled in the configuration of this integration by setting power_control to true. When power control is enabled number.solaredge_active_power_limit is available in Home Assistant for reading and writing. The actual value of the active power limit is read together with the other values of the inverter.

Now I only need to figure out where I can put this config line...

gijsrrr, 2024-05-01:
> Isn't this related to this remark in the documentation? Power control is disabled by default. It can be enabled in the configuration of this integration by setting power_control to true. When power [...]

You need to put it in 
`.storage\core.config_entries`

Mine looks as such (I have power control set to TRUE)
`      {
        "entry_id": "<REDACTED>",
        "version": 1,
        "minor_version": 1,
        "domain": "solaredge_modbus",
        "title": "solaredge",
        "data": {
          "name": "solaredge",
          "host": "<REDACTED>",
          "port": 1502,
          "modbus_address": 1,
          "read_meter_1": false,
          "read_meter_2": false,
          "read_meter_3": false,
          "read_battery_1": false,
          "read_battery_2": false,
          "power_control": true,
          "scan_interval": 30
        },`

I'm still not able to adjust the power control entry, this did work before.

Stievo9997, 2024-05-02:
I did followed : https://github.com/binsentsu/home-assistant-solaredge-modbus?tab=readme-ov-file#enabling-power-control-on-solaredge-inverter

So i enabled Reactive power mode to RRCR, but it is still not working.
The only thing which i couldn't find is Power Control to enable.

But the strange thing is, that it worked before,

I found someone with the same issue: https://community.homey.app/t/app-pro-solaredge-growatt-tcp-modbus/45487/945 he enabled all the possible things in the inverter. After that it worked, but dont know what the specfic setting is to make this work again

mourmeer, 2024-05-21:
where do I find:  .storage\core.config_entries   ??

biemond, 2024-05-25:
looks like some settings are sticky or changed after something was done on the solaredge mobile app. 

This user had success by doing a reset of the inverter.

https://github.com/biemond/solaredge.modbus/issues/73#issuecomment-2123489897

infoedkok, 2024-09-03:
Hi, I am using solaredge modbus integration on a docker in Synology and it works fine with me! 
I can change "solaredge (Active Power Limit)" manually from e.g 100%to 50%. And after some seconds, I see an actual decrase  of sun energy production . Worked with different versions of HA and the integration.

ONE TIP: perhaps you should check your access rights to your SolarEdge installation. You can only change such inverter paramaters if you are the "system owner". If not, ask solar edge support or your SE dealer.

Goran129, 2025-03-19:
Were do I see if I am the system owner?

infoedkok, 2025-03-20:
You know it when you cannot set the settings in your app. I am a dutch user so I see the dutch words. On youtube, there are instruction videos. 

I noticed I could not use these settings, so I contacted solarsupport. They switched my account from "system user" to "system owner". They saw that my installer of my system gave me to less rights to the system settings. After Solaredge support made me a system owner, I could use the settings in the app.

Hope this works for you also!

Goran129, 2025-03-20:
Well I can change settings in my SolarEdge app. Seem to be something else wrong then. Or do you mean change settings in the SetApp? There I log in with another account.

janprive, 2025-12-07:
> You know it when you cannot set the settings in your app. I am a dutch user so I see the dutch words. On youtube, there are instruction videos.
>
> I noticed I could not use these settings, so I [...]

Hi! can you tell me how you contacted solarsupport etc to change the account? I want to haved changed it too!