binsentsu/home-assistant-solaredge-modbus #234: Home Assistant web interface stops working after inverter network connection is down

romke81, 2024-05-15:
Hello, the solaredge modbus works most of the time perfectly. But in case of an network connection error on the solaredge inverter the webinterface from Home assistant stops working. When i reboot home assistant via CLI i have 2 minutes to disable the solaredge modbus integration. After the 2 minutes home assistant web interface crashes again. Why Home assistant stops when the solaredge modbus integration cannot get any data from the inverter. This issue does not exist when the inverter goes to sleep mode.

Timon321, 2024-08-19:
I am having the same problem with another integration of solaredge-modbus (https://github.com/erikarenhill/solaredge-modbus-hass).
And i dont know, how to debug this thing.

Hooglander1, 2024-10-04:
Just experienced the exact same issue the last two days. Pulled my hair out what could be the issue, but it ended up being this integration that somehow pulled the whole web interface down. In my case the inverter switched IP addresses and changing that back manually to the previous fixed it immediately. 

In the logs of HA core, the error message pops up every second, but maybe in the backend there are more retries maybe?