binsentsu/home-assistant-solaredge-modbus #240: async_forward_entry_setup deprecation

brembygit, 2024-07-08:
Detected that custom integration ‘solaredge_modbus’ calls async_forward_entry_setup for integration, solaredge_modbus with title: solaredge and entry_id: 7ba09082a3eb081f56ec1643a9fa2fd8, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/solaredge_modbus/init.py, line 130: hass.async_create_task(, please report it to the author of the ‘solaredge_modbus’ custom integration

Detected code that calls async_forward_entry_setup for integration solaredge_modbus with title: solaredge and entry_id: 7ba09082a3eb081f56ec1643a9fa2fd8, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.

nagtegaal, 2024-08-14:
Duplicate of [issue 239](https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/239)

binsentsu, 2024-08-23:
fixed in [V1.12.0](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.12.0)

closed 2024-08-23 by binsentsu