binsentsu/home-assistant-solaredge-modbus #247: async_forward_entry_setup, which can cause the setup lock

Erwiiinst, 2024-08-12:
Logger: homeassistant.helpers.frame
Bron: helpers/frame.py:151
Eerst voorgekomen: 22:36:38 (5 gebeurtenissen)
Laatst gelogd: 22:36:43

Detected code that calls async_forward_entry_setup for integration solaredge_modbus with title: solaredge and entry_id: 547c5966f3e97a81f8046cf9479fd39a, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.

nagtegaal, 2024-08-14:
Duplicate of [issue 239](https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/239)

binsentsu, 2024-08-23:
fixed in [V1.12.0](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V1.12.0)

closed 2024-08-23 by binsentsu