binsentsu/home-assistant-solaredge-modbus #253: SolarEdge Modbus entities unavailable

Scigik, 2024-08-23:
Hi, since 8 Aug 2024, apparently around at 0.00 am it seems my SolarEdge modbus entities became unavailable. The Energy Dashboard and all automations depending on the SolarEdge Modbus entities are now non-functional.

[home-assistant_solaredge_modbus_2024-08-23T19-57-41.341Z.log](https://github.com/user-attachments/files/16733628/home-assistant_solaredge_modbus_2024-08-23T19-57-41.341Z.log)
 I'm new to HA, so not entirely sure how to troubleshoot this. I see in the backup log that HA was updated from core 2024.6.4 to version 2024.7.3 on 9 Aug, so after the start of the bug. The SolarEdge Modbus integration is on version 1.11.1.

I tried to run a debug log on the SolarEdge Modbus entity today (see attached) and saw a few potential issues without knowing how these could have been caused:

"[homeassistant.helpers.frame] Detected that custom integration 'solaredgemodbus' calls asyncforward_entry_setup for integration, solaredge_modbus with title: solaredge and entry_id: 452a94259733555d8ec1f30087c43d3a, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/solaredge_modbus/__init.py, line 130: hass.async_create_task(, please report it to the author of the 'solaredge_modbus' custom integration"

"2024-08-23 15:53:00.723 WARNING (SyncWorker_0) [custom_components.solaredge_modbus] not able to connect to 192.168.50.149:1502
2024-08-23 15:53:15.683 ERROR (SyncWorker_23) [pymodbus.logging] Connection to (192.168.50.149, 1502) failed: [Errno 113] Host is unreachable"

Could you please advise?

Thanks a lot!

robi59italy, 2024-08-24:
update HA to 2024.08.02

Scigik, 2024-08-24:
That's already the case. I also just applied the new update to the SolarEdge Modbus integration (now on 1.12.0) but the problem remains.

Any other ideas?

Thank you

Scigik, 2024-08-26:
[image]

Scigik, 2024-08-26:
[image]
[image]

Scigik, 2024-08-26:
I also captured an updated logfile, hoping it can help you assess what's not working. 
[home-assistant_solaredge_modbus_2024-08-26T19-56-04.804Z.log](https://github.com/user-attachments/files/16753120/home-assistant_solaredge_modbus_2024-08-26T19-56-04.804Z.log)

Thanks a lot for the support.

Kindly

robi59italy, 2024-08-26:
are you sure your inverter is connected? Try to ping it.

Scigik, 2024-08-26:
The ping seems successful:

Pinging 192.168.50.149 with 32 bytes of data:
Reply from 192.168.50.104: Destination host unreachable.
Reply from 192.168.50.104: Destination host unreachable.
Reply from 192.168.50.104: Destination host unreachable.
Reply from 192.168.50.104: Destination host unreachable.

Ping statistics for 192.168.50.149:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),

robi59italy, 2024-08-26:
no it is not connected you should get time in millisecond...
[image]

Scigik, 2024-09-01:
@robi59italy Update: seems that something changed in the inverter configuration (not my doing, presumably SolarEdge?) which changed the IP address of the inverter, which is why the connection wasn't working. Thanks for your support!

closed 2024-09-01 by Scigik