binsentsu/home-assistant-solaredge-modbus #260: Integration nog getting data after update to HA 2025.1

RubberDuckNL, 2025-01-04:
Hi all, after updating my home assistant to 2025.1 I do not get any data through the modbus integration anymore. For confirmation I restored the backup from before (2024.12) and then I instantly get my data visible again. Anyone who can look into it? 

I am not at all an expert in this kind of programming, but I managed to get this from the logs (I hope it can be of any use):
2025-01-04 10:30:38.267 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'solaredge_modbus' calls async_forward_entry_setup for integration, solaredge_modbus with title: solaredge and entry_id: 5b83ef322a32cb0c6407fc5d63296320, which is deprecated, await async_forward_entry_setups instead at custom_components/solaredge_modbus/__init__.py, line 125: hass.async_create_task(. This will stop working in Home Assistant 2025.6, please report it to the author of the 'solaredge_modbus' custom integration

wanted156, 2025-01-04:
I have experienced the same issue, but I've rolled back my HA instance already (without saving the logs...). I can update again and retrieve more logs if necessary.

formiification, 2025-01-04:
same issue here

hvraven, 2025-01-04:
same here. According to https://github.com/home-assistant/core/issues/134593 it is a version conflict between the pinned pymodbus in this integration and the updated version in home assistant 2025.1

binsentsu, 2025-01-04:
V1.13.0 was released targeting HA 2025.1 with updated pymodbus dependency. This should resolve conflict.

RubberDuckNL, 2025-01-04:
Can confirm, it works now. Stupid me, didn't realise you could update the hacs integrations... Thanks!

steppige, 2025-01-05:
I don't know why but it doesn't work for me.. I updated to V1.13.0 and I still don't receive data from my solaredge, do you recommend doing a Restore to my previous backup?

tias2001, 2025-01-06:
Hi
After updating, communication is working on and off, sometimes no data received for 3 hours.  Reloading integration gets it updating again but drops off again a few hours later..
Is anyone else experiencing this behaviour?

jgarridoalcazar, 2025-01-06:
I am also experiencing the same issue. HA 2025.1 with SE Modbus v1.13 results in disconnecting the communication and wrong values in many entities. The log shows errors in the structure of modbus data values (missing bytes). Seems something related with the pymodbus library.

In my case, going back to Modbus v1.12 still works fine with HA 2025.1.

gndean, 2025-01-08:
I have the same issue as @tias2001 and @jgarridoalcazar, HA 2025.1.1, SE Modbus 1.13.0. Sensors not updating reliably. The logs contain many instances of this:
```
Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 320, in read_modbus_data
    and self.read_modbus_data_meter1()
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 331, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 356, in read_modbus_data_meter
    accurrenta = decoder.decode_16bit_int()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 397, in decode_16bit_int
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes
```

markopen, 2025-01-09:
With the last update, I'm also having trouble getting the power plant output data. When I restart the system, the data is visible again...

[image]

essendyx, 2025-01-13:
same problem here.

markopen, 2025-01-13:
Even the latest HA Core 2025.2 update does not solve the problem with updating data from Solaredge Modbus. When I restart the system, this data is refreshed again....

binsentsu, 2025-01-13:
If you don't use the HA standard modbus integration for any other modbus sensors, the pymodbus version defined in this integrations' manifest is loaded. If this is the case,you can safely downgrade this integration to V1.12 (uses pymodbus 3.6.9). Check if the sensor updating  problem is then resolved. There were some changes in the external modbus library used by HA and this integration for decoding the binary payloads which may have impact on the output.

markopen, 2025-01-17:
With downgraded version Solaredge modbus V1.12  problems not resolved... :(