binsentsu/home-assistant-solaredge-modbus #261: Error reading modbus data (OverflowError: int too large to convert to float) - Spamming since 2025.1.0

w1Ngx, 2025-01-09:
Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:244
integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: January 8, 2025 at 11:02:53 PM (662 occurrences)
Last logged: 9:08:27 AM

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 318, in read_modbus_data
    self.read_modbus_data_inverter()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 686, in read_modbus_data_inverter
    self.data["acenergy"] = round(acenergy * 0.001, 3)
                                  ~~~~~~~~~^~~~~~~
OverflowError: int too large to convert to float

pcdiks, 2025-01-21:
I have this error as of today. What changed? I updated the firmware on my SolarEdge.
Logs:
`
2025-01-21 20:15:03.280 ERROR (SyncWorker_7) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 321, in read_modbus_data
    and self.read_modbus_data_meter2()
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 336, in read_modbus_data_meter2
    return self.read_modbus_data_meter("m2_", 40364)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 471, in read_modbus_data_meter
    exporteda = decoder.decode_32bit_uint()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 375, in decode_32bit_uint
    return unpack("!" + fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes
    accurrenta = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 367, in decode_16bit_uint
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 686, in read_modbus_data_inverter
    self.data["acenergy"] = round(acenergy * 0.001, 3)
                                  ~~~~~~~~~^~~~~~~
OverflowError: int too large to convert to float 
`
Old vs new firmware versions:

[image]

BerHa1971, 2025-02-13:
Hi @pcdiks and team.

Since today noon I do have the same issue like you have "int too large to convert to float". I did not do any updates manually, but the CPU version (4.22.44) is exactly the same version as yours. I assume (wasn't aware of the functionality) that an auto update did happen on my side?

BTW: Any ideas why some of my software versions are way older than yours? I did just connect with the SolarEdge Setapp software (it did download a firmware update when opening the app), but there were not updates applied to my SolarEdge Inverter (SE8K-RWS48BEN4).

[image]

[image]

Thanks in advance!

BerHa1971, 2025-02-13:
Update on this issue. 

I am using modbus-proxy since a couple of days (and used it successfully before), because I am also trying to leverage evcc to optimize my car charging experience, and after a restart of home assistant (this includes restarting modbus-proxy, the solaredge integration and all the rest) I still had the same issue.

Now I restarted modbus-proxy a couple of times (not the full home assistant instance) and now **the error disappered**. Maybe the error message "int too large to convert to float" is missleading and the actual error was a **connection issue** between Home Assitant solaredge modbus and the modbus proxy / the inverter?

Thanks

Cengiz87, 2025-12-06:
The issue occurs because the SolarEdge Modbus integration receives an extremely large integer value from a sensor (`m1_importvarhq3`), in my case  

- When Home Assistant tries to convert this value to a float internally, it causes an `OverflowError: int too large to convert to float`.  
- The root cause is that the SolarEdge inverter (or Modbus register mapping) is sending a value that is far larger than expected.  
- This is not a bug in Home Assistant itself, but the integration currently does not handle such unrealistic values safely.  
- Logs show debug output like:

```
2025-12-06 17:53:47.021 DEBUG (MainThread) [custom_components.solaredge_modbus.sensor] Solaredge sensor 'm1_importvarhq3': raw new_value = 4600632200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

Cengiz87, 2025-12-06:
I changed te code (in sensor.py of custom_component row 90-94) to prevent to high numbers are given by solaredge modbus which cant be handled

```
        if (
            (self.entity_description.state_class != SensorStateClass.TOTAL_INCREASING)
            or (new_value is None)
            or (new_value > 0 and new_value < 1_000_000_000_000)
        ):