binsentsu/home-assistant-solaredge-modbus #262: Error reading modbus data (struct.error: unpack requires a buffer of 2 bytes) - Spamming since 2025.1.0

w1Ngx, 2025-01-09:
Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:244
integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: January 8, 2025 at 11:02:41 PM (669 occurrences)
Last logged: 9:08:18 AM

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 320, in read_modbus_data
    and self.read_modbus_data_meter1()
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 331, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 356, in read_modbus_data_meter
    accurrenta = decoder.decode_16bit_int()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 397, in decode_16bit_int
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes

guntherd, 2025-01-27:
I can confirm same issue and strange behavior. I have updated to Solaredge modus 2.0.0 / HA OS 14.1 / Core 2025.1.4
Look at the short attached video how this looks on the frontend and the attached logfile for my case.

[solaredge-modbus.log](https://github.com/user-attachments/files/18558718/solaredge-modbus.log)
[image]

Before the upgrade I already had issues. The complete reading of modbus can be down for half a day or longer and then it starts working again. Now with v2.0.0 the behavior has changed as describe above.

binsentsu, 2025-01-27:
@guntherd Thanks for the log. Is really helpful. There was an error regarding the battery status, this was introduced in 2.0.0. This should be fixed in 2.0.1. The unpack error already started occurring within V1.x release, after upgrade to HA core 2025.1. In that release the underlying pymodbus library was updated from 3.6.9 to 3.7.4 within HA Core in which a refactoring happened in the decoder class. To rule this out, I now use again the decoder of 3.6.9 locally within this project to see if this unpack error resolves then. Keep me posted if this fixes issue for you.

guntherd, 2025-01-30:
Yesterday I thought the issues are fixed but today I have again issues. Note: the manifest still shows version 2.0.0, but I have installed version 2.0.1 and updated the HA OS. I again attach my logfile.

[home-assistant_solaredge_modbus_2025-01-30T19-52-12.264Z.log](https://github.com/user-attachments/files/18608032/home-assistant_solaredge_modbus_2025-01-30T19-52-12.264Z.log)