binsentsu/home-assistant-solaredge-modbus #265: SolarEdgeModBus don't work since 2025.1.2

nuki47, 2025-01-12:
Since the update I don't get any data in the solaredge-modbus integration (all sensors state: unknown)

Core 2025.1.2
Home Assistant OS
solaredge-modus 1.13.0

That's all in the logfile:
```
2025-01-12 16:11:20.103 DEBUG (MainThread) [custom_components.solaredge_modbus] Setup solaredge_modbus.solaredgemod
2025-01-12 16:11:38.808 INFO (SyncWorker_8) [custom_components.solaredge_modbus] modbus client is not connected, trying to reconnect
2025-01-12 16:11:38.809 INFO (SyncWorker_8) [custom_components.solaredge_modbus] successfully connected to 192.168.100.179:502
```

s50u, 2025-01-13:
Exactly the same for me.
Core 2025.1.2
Supervisor 2024.12.3
Operating System 14.1
solaredge-modus 1.13.0

markopen, 2025-01-14:
Latest HA Core 2025.2 update does not solve the problem with the occasional stoppage of data updates from Solaredge Modbus. When I restart the system or reload integration, this data is refreshed again....

markopen, 2025-01-20:
Can we expect a solution to the problem with frequent non-functioning of the latest version of the Solaregda Modbus extension?

markopen, 2025-01-21:
HA Core update to ver. 2025.1.3. - No changes! SolarEdge Modbus frequent non-functioning ...

grouts112, 2025-01-23:
While Waiting for a fix to the daily  ERROR stopping this not working,
I have made an automation to trigger on the error and restart the integration.
Working nicely.

grouts112, 2025-01-23:
Looking at https://github.com/pymodbus-dev/pymodbus/issues?q=is%3Aissue%20state%3Aclosed

it seems some similar issues reported against pymodbus 3.7.4  which is what was updated from pymodbus 3.6.9 in the recent solareedge_modbus update to fix the 2025.1 issues.
  From what I can see (in my poor understanding) this needs updating to at least pymodbus 3.8.0 ?

grouts112, 2025-01-23:
Seems pymodbus 3.8.3 is merged into core.
https://github.com/home-assistant/core/pull/134809

I can't tell what release that will be in.  2025.1.4 ??
Maybe that will fix this.

markopen, 2025-01-23:
> While Waiting for a fix to the daily ERROR stopping this not working, I have made an automation to trigger on the error and restart the integration. Working nicely.

Can you share this automation?
Thanks a lot

grouts112, 2025-01-23:
here is the Automation:
   Note... you will need to find your specific "entry_id" for your solaredge_modbus integration on your HA.
Check in the 'config/.storage/config_entries' file...  search for "solaredge_modbus" and copy the long number after 'entry_id'
 and replace my number

`alias: Solar - restart modbus if syslog error
description: ""
triggers:
  - trigger: event
    event_type: system_log_event
    event_data:
      level: ERROR
      name: custom_components.solaredge_modbus
conditions:
  - condition: template
    value_template: "{{ 'Error reading modbus data' in trigger.event.data.message }}"
actions:
  - data:
      entry_id: 807771735aa13ef5ef4f3d47e0bfeb72
    action: homeassistant.reload_config_entry
    target:
      device_id: 8dadc9e485a9ad346f265ea2b32b62d8
  - action: persistent_notification.create
    data:
      title: restarted solaredge_modbus service
      message: solaredge_modbus failed to read
    enabled: true
`

markopen, 2025-01-25:
Followed the instructions, found the device ID, replaced the number. Triggering the automation reports an error: 

[image]

Automation YAML:

alias: Solar - restart modbus
description: ""
triggers:
  - trigger: event
    event_type: system_log_event
    event_data:
      level: ERROR
      name: custom_components.solaredge_modbus
conditions:
  - condition: template
    value_template: "{{ 'Error reading modbus data' in trigger.event.data.message }}"
    enabled: false
actions:
  - data:
      entry_id: 3b5133d3283c41bfddf666043798db7c
    action: homeassistant.reload_config_entry
    target:
      device_id: 8dadc9e485a9ad346f265ea2b32b62d8
  - action: persistent_notification.create
    data:
      title: restarted solaredge_modbus service
      message: solaredge_modbus failed to read
    enabled: true

grouts112, 2025-01-25:
I suspect that your ‘target’ device id: will need changing also.
When in the automation editor, (not yaml view)
 Expand the action and change the device to pick your solaredge inverter,
That’s what I did, I didn’t think about that bit and that it would be different for yours, sorry.

[image]

grouts112, 2025-01-25:
Ooh just re read your post..
 That is the message i put in the automation, so you get a notification that is did it, otherwise you wouldnt know.
You could delete / disable that “notify” action at the bottom of the automation

binsentsu, 2025-01-25:
Was running V2.0 already some time at my house, which handles updates differently. Did not have any drop outs on HA 2025.1.x. Pushed the release out now. It's not really clear to me if the integration just stops updating its data without errors, or everybody has the modbus unpack error mentioned after which no updates are coming in anymore.

grouts112, 2025-01-25:
Thank you @ binsentsu

markopen, 2025-01-27:
After 24 hours with latest updates I got this....

[image]

binsentsu, 2025-02-01:
In 2.0.2 the modbus connection is re-initiated automatically upon modbus read errors of the underlying modbus library.