binsentsu/home-assistant-solaredge-modbus #267: after update HA to 2025.01.03 solaredge modbus stopped to work

juzis28, 2025-01-23:
updated homeassistant to 2025.01.03 and integration stopped to work. All solaredge entities are unknown. Also show error: "Entity is non numeric"

markopen, 2025-01-25:
HA Core update to ver. 2025.1.4. - No changes! SolarEdge Modbus frequent stopped to work ...

TheLordVader, 2025-01-26:
I can confirm the same isue.

markopen, 2025-01-26:
> I can confirm the same isue. Reloading the integration causes starts modbus to work again.

Can you upgrade Solaredge Modbus to ver. 2.0 ?

TheLordVader, 2025-01-26:
Yes I did. Problem persists.

markopen, 2025-01-27:
I confirm....

[image]

TheLordVader, 2025-01-27:
@binsentsu are you aware of this issue?

binsentsu, 2025-01-27:
Thing is, I can't reproduce it. I'm running version 2.0 without issues. 
There are variables which influenced the communication in the past, like running wired ethernet or through wifi and also cases where specific firmware versions of solaredge bricked the communication.
I'm only running an inverter, so no meters or batteries. This on solaredge firmware 4.21.23 of wired ethernet.
In order to get more grip on these issues such details are important.

markopen, 2025-01-27:
@binsentsu Hello. I am using a SolarEdge inverter with firmware version 4.22.44. The data is transmitted via a wired LAN connection. I do not have any battery storage systems. However, I have an Inepro Pro380 meter connected for measuring energy import and export. Until a few firmware versions ago, the same configuration worked without any issues for almost two years...

TheLordVader, 2025-01-27:
@binsentsu thanks for answering. I have a Solaredge 15k connected via ethernet cable. No battery systems but a M1 (solaredge) smart meter, Firmware is updated to the latest version last week through SetApp, I can check version if that is something you would like to know.

binsentsu, 2025-01-27:
@TheLordVader yes, please check firmware version. Will probably be also version 4.22

markopen, 2025-01-28:
Updated the integration to version 2.0.1. Integration still shows version 2.0.0...

[image]

binsentsu, 2025-01-28:
Forgot to bump version in the separate manifest file. It's 2.0.1 if hacs mentioned it like that upon upgrade.

dvoijen, 2025-01-28:
Version 2.0.1 still sets causing the constgant unknown issues.

dcseek, 2025-01-28:
Version 2.0.1 same here no data from modbus via wifi

TheLordVader, 2025-01-28:
@binsentsu here are my firmware versions. Modbus stopped working again, I was hopeful when it worked for several hours, but nope...

[image]

ItsMe00007, 2025-01-28:
I am using HA 2025.1.4 and Modbus TCP 2.0.1 and I am getting frequently the following error message in the system Log and no values at the gui at that time.
After a Restart of the HA it seems to work for some time.

This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: helpers/update_coordinator.py:411
integration: SolarEdge Modbus (documentation)
First occurred: 22:07:04 (24 occurrences)
Last logged: 22:30:04

Error fetching solaredgeSE10k data: Error updating modbus data: unpack requires a buffer of 2 bytes

binsentsu, 2025-01-28:
@ItsMe00007 can you list the firmware version of your inverter and what devices (inverter,battery,meter) u use?

markopen, 2025-01-29:
After 24 hours with version 2.0.1 Solaredge modbus integration still working without problems...

ItsMe00007, 2025-01-29:
> [@ItsMe00007](https://github.com/ItsMe00007) can you list the firmware version of your inverter and what devices (inverter,battery,meter) u use?

I have only an SE10K Inverter.
Here‘s my Firmware:

[image]

ItsMe00007, 2025-01-29:
Today the Buffer errors appeared again after yesterday's HA Restart:

This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: helpers/update_coordinator.py:411
integration: SolarEdge Modbus (documentation)
First occurred: 14:52:24 (94 occurrences)
Last logged: 19:19:11

Error fetching solaredgeSE10k data: Error updating modbus data: unpack requires a buffer of 2 bytes

ItsMe00007, 2025-02-01:
> [@ItsMe00007](https://github.com/ItsMe00007) try out https://github.com/WillCodeForCats/solaredge-modbus-multi it seems to work very good

Thanks for the tip, I'll give it a try. 
But this integration was working very well until lately, I just installed the new update now and I am hoping that it will fix the problem.

TheLordVader, 2025-02-01:
@ItsMe00007 yes, I have bben using it for many years and it has worked flawlessly. I have now converted into the other "solaredge modbus multi" and I have to say it's even better. Separate devices for inverter and meter, plus many more addidtional features. So far I like it very much! And, of course, it is working...

ItsMe00007, 2025-02-01:
> [@ItsMe00007](https://github.com/ItsMe00007) yes, I have bben using it for many years and it has worked flawlessly. I have now converted into the other "solaredge modbus multi" and I have to say [...]

I just installed it, looks good, especially if one uses battery and meter.

For the moment I stick to the single inverter implementation since I have quite some automations which I would also have to change.

But thanks for the great tip.

ItsMe00007, 2025-02-03:
I also just  checked the "Active Power Limit" History and it was always 100% since I never changed it.
But since January 11th the values got crazy, see screenshot.
I guess that I installed the HA 2025.1.2 at that day, but not sure.

[image]

ItsMe00007, 2025-02-05:
Still the same buffer problem with the current version:

This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: helpers/update_coordinator.py:411
integration: SolarEdge Modbus (documentation)
First occurred: 01:23:00 (5 occurrences)
Last logged: 21:05:48

Error fetching solaredgeSE10k data: Error updating modbus data: unpack requires a buffer of 2 bytes

ItsMe00007, 2025-03-12:
Still the same Buffer Problem with 2.0.3 and HA 2025.03.02

This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: helpers/update_coordinator.py:412
integration: SolarEdge Modbus (documentation)
First occurred: 16:09:47 (9 occurrences)
Last logged: 21:10:46

Error fetching solaredgeSE10k data: Error updating modbus data: unpack requires a buffer of 2 bytes
Error fetching solaredgeSE10k data: Error updating modbus data: [Errno 104] Connection reset by peer
Error fetching solaredgeSE10k data: Error updating modbus data: Modbus Error: [Connection] ModbusTcpClient 192.168.50.140:1502: Connection unexpectedly closed 0.000 seconds into read of unbounded read bytes without response from slave before it closed connection