binsentsu/home-assistant-solaredge-modbus #274: Repair with state class after upgrade to 2.0.0

vadius45, 2025-01-26:
Hello,

After the upgrade to 2.0.0 , I had to repair a bunch of entities due to the change of the unit, no big deal. 
And I'm left with several errors such as: 

The entity no longer has a state class
We have generated statistics for 'modbus M1 EXPORTED A VAh' (sensor.solaredge_m1_exported_a_vah) in the past, but it no longer has a state class, therefore, we cannot track long term statistics for it anymore.

Statistics cannot be generated until this entity has a supported state class.
If the state class was previously provided by an integration, this might be a bug. Please report an issue.
If you previously set the state class yourself, please correct it. [The different state classes and when to use which can be found in the developer documentation.](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)
If the state class has permanently been removed, you may want to delete the long term statistics of it from your database.
Do you want to permanently delete the long term statistics of sensor.solaredge_m1_exported_a_vah from your database?

What should I do ? Is it something that can be fixed (a bug ?) or something that is no longer available ? 

Thanks

boecks, 2025-01-27:
Hi, same here. Justed clicked through and took the suggested options...
#273

binsentsu, 2025-01-27:
Check 2.0.1 update for fix of VAh and varh sensors

closed 2025-01-28 by vadius45