binsentsu/home-assistant-solaredge-modbus #275: Upgrade to 2.0 Doesn't See 2nd Battery and has other issues.

TDStrasser, 2025-01-27:
After upgrading to 2.0 several errors and issues occurred:
1. Issues - the meter units changed from "VAR" to "var" throwing up a lot of issues to fix (also reported by others).
2.  Battery status became unavailable on Battery 1
3.  Battery 2 became unavailable in its entirety.
4.  The appearance of the control parameter entities became scrambled with some wide and some narrow button blocks.

I then deleted the device and reconfigured my system afresh ensuring I had selected the correct IP address, Modbus address, number of meters and number of batteries. All to no avail as the issues above persisted. I also tried a restart, still the issues remained.

In the end I rolled back to V1.19, reinstalled and configured, and the device and entities came back as normal.

It looks like V2.0 needs some more work before it's ready to use for those with demand meters and batteries.

binsentsu, 2025-01-27:
hi,

1.  Integrations are encouraged to use the official units as defined in HA core. This integration was migrated towards them. It used to be self defined with lowercase value. The repair will be necessary. Hence, these changes the major version was bumped.
2. Should be fixed in 2.0.1
3. Could not yet identify this issue. Provide logs if any.
4. Not clear what you mean. Integration does not render ui itself.

eikejung, 2025-01-28:
Hi @binsentsu,
thanks for your reply. When do you expect the release of 2.0.1? I'm thinking about downgrading, but would like to avoid it... ;)

binsentsu, 2025-01-28:
Released it yesterday

eikejung, 2025-01-28:
I'm sorry - didn't see that. At the moment 2.0 is provided in HA - 2.0.1. not yet...

TDStrasser, 2025-01-28:
Thanks for the responses.
I'll try 2.0.1 when I get some time. For now the rolled back version is working fine for me.
I looked for logs from the brief period V2.0 was on my system. The attached file is an extract that seems to throw up the related errors.

[HASS TCP Modbus Errors.txt](https://github.com/user-attachments/files/18574301/HASS.TCP.Modbus.Errors.txt)