binsentsu/home-assistant-solaredge-modbus #277: Enable setting active power limit for existing device

pascalwinters, 2025-02-03:
For an existing device with the active power disabled, there is no way to enable it afterwards through the UI.
Is it possible to make a reconfigure option (like the Shelly integration), to enabled it afterwards?

Example Shelly:

[image]

Jay-El66, 2025-02-03:
After the 2.0.2 update (or the 2.0.1, which behaved really strange), the whole entity "active power limit" seems gone:

_This entity is no longer being provided by the solaredge_modbus integration. If the entity is no longer in use, delete it in settings._

I didn't see that in te notes, but maybe I missed it.

Ang3rst, 2025-02-09:
Same here entity "number.solaredge_active_power_limit" was grayed-out, and was not sure when that happend.
So did looked into some code changes and did see a lot of edits here on that value here (re-write code if im correct)
https://github.com/binsentsu/home-assistant-solaredge-modbus/commit/4fe2f686abf3cd811b34467612181c4b9e4783a4

The release before that one was V1.13.0.
So downgrade the "Solaredge Modbus" version in HACS to V1.13.0, and all is good again. (did try v2.0, no luck)
I'll wait for an fix (if any) but for now downgraded and ignored lastest version

_In 'Home assistant' go to HACS
click on Solaredge Modbus addon tp open the addon page
Top right click on the 3 dots
Select Re-download
Choose your version under "Need a different version?"_

**HW
Inverter: SolarEdge SE3000H

Jay-El66, 2025-02-09:
> 
> The release before that one was V1.13.0. So downgrade the "Solaredge Modbus" version in HACS to V1.13.0, and all is good again. (did try v2.0, no luck) I'll wait for an fix (if any) but for now downgraded and ignored lastest version

Thanks, did the same, roll back to 1.13.0 and it's back indeed.

So hoping the code will be fixed soon.

edgarvoorbraak, 2025-02-09:
You can add to existing device by editing core.config_entries and restarting HA:
 _"power_control":true_ 
to the line after; 
_"port":1502,_
take care to properly separate the arguments by commas.

Ang3rst, 2025-02-10:
@edgarvoorbraak
Thanks that worked, back on stable v2.0.2

FIrst I was unable to see the 'core.config_entries' file in '/homeassistant/.storage' in File Editor (Raspberry Pi setup).
After some reading found out I needed to remove the .config "Ignore Pattern" from File Editor Setup. (setting) to be able to see that folder.

_In 'Home assistant' go to Setting > Addons > File editor > configurator
Press the X (remove) behind .config
and  SAVE.
Now in File Editor the folder (and files) are shown._

then changed the value as u discribed above. " "power_control":true"

Jay-El66, 2025-02-25:
Why wasn't this fixed in 2.0.3?
I Don't want to manually change config files only for them to be overridden again by an update.
Please re-enable power control

edgarvoorbraak, 2025-02-25:
The fix I did in 2.0.2 still holds so no override being done .

Jay-El66, 2025-02-26:
> The fix I did in 2.0.2 still holds so no override being done .

That's good to know, thanks for the feedback. I now applied the change to the config file which indeed reinstates the power-control option. 
Thanks again.

github0123456789, 2025-03-12:
> @edgarvoorbraak
> Thanks that worked, back on stable v2.0.2
> 
> FIrst I was unable to see the 'core.config_entries' file in '/homeassistant/.storage' in File Editor (Raspberry Pi setup).
> After some reading found out I needed to remove the .config "Ignore Pattern" from File Editor Setup. (setting) to be able to see that folder.
> 
> _In 'Home assistant' go to Setting > Addons > File editor > configurator
> Press the X (remove) behind .config
> and  SAVE.
> Now in File Editor the folder (and files) are shown._
> 
> then changed the value as u discribed above. " "power_control":true"
> Reboot

Thank you. Your guidance saved my day. I initially thought and found false information this all had to do with a firmware update at my inverter.