binsentsu/home-assistant-solaredge-modbus #278: Unable to set export control limit number greater than 0

Ralphy140, 2025-02-10:
Version 2.0.2

Since updating it seem that the ability to edit the export control site limit has been broken. When trying to set with a slider it will no go above 0 and if I run using the set.number for any value greater than 0 I get the below error with some sort of buffer overflow and the value going to -3 trillion 

Value 5000.0 for number.solaredge_export_control_site_limit is outside valid range 0 - False

This prevents me from being able to curtail solar when power prices are negative in the power marker.

Holguras, 2025-04-04:
This happens to me too, but I'm not sure if it's because of the latest inverter update. I've tried downgrading to Solaredge Modbus version V1.X.X and the same thing happens.

RacingHell, 2025-05-21:
ChatGPT mentions:

Note on Home Assistant Core 2024.5+ Compatibility:

Since updating to Home Assistant Core 2024.5, the number.solaredge_export_control_site_limit entity has become non-functional for automations and UI interactions. Previously, it accepted manual values without issues. Now, attempting to set a value results in the following error:

```yaml
Value 6070.0 for number.solaredge_export_control_site_limit is outside valid range 0 - False
```

This indicates that Home Assistant interprets the max value of the number entity as False, rendering any value above 0 invalid.

Cause:

The issue arises because the entity lacks an explicitly defined max value. In Home Assistant 2024.5 and later, number entities must have valid min and max values defined. Without these, Home Assistant defaults to max = False, effectively preventing any value from being set.

Proposed Solution:

To resolve this, please update the Modbus number entity definition for Export Control Site Limit (address: 40178) to include explicit min and max values. For example:

```yaml
- name: Export Control Site Limit
  address: 40178
  slave: 1
  input_type: holding
  unit_of_measurement: "W"
  data_type: uint16
  scale: 1
  offset: 0
  precision: 0
  device_class: power
  min: 0
  max: 10000  # Adjust according to your inverter's specifications
```

This adjustment will ensure compatibility with Home Assistant’s updated requirements and restore full functionality to the entity.