binsentsu/home-assistant-solaredge-modbus #280: SolarEdgeModBus don't work

PabloFFF93, 2025-02-10:
Hello, I have 3 inverters and everything worked fine until now.

One day 2 out of 3 stopped communicating. Data reaches the Solaredge cloud.

After deleting and adding I get the message "Unknown error occurred"
After some time I can add it but HA shows in the logs "No response received after 3 retries, continue with next request"

1 inverter works, communicates correctly

blowk, 2025-02-10:
What error do you see in the log files?
I get a couple of times, during the day, an error like :

`Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 318, in read_modbus_data
    self.read_modbus_data_inverter()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 684, in read_modbus_data_inverter
    acenergy = validate(self.calculate_value(acenergy, acenergysf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 160, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)`

PabloFFF93, 2025-02-10:
Ten błąd pochodzi z niestandardowej integracji.

Rejestrator: pymodbus.logging
Źródło: custom_components/solaredge_modbus/__init__.py:291
integracja: SolarEdge Modbus (dokumentacja)
Pierwsze zdarzenie: 21:52:09 (1 zdarzenia)
Ostatnio zalogowany: 21:52:09

No response received after 3 retries, continue with next request

blowk, 2025-02-11:
Can you tell which version of Solaredge Modbus and Home Assistant you are using?

PabloFFF93, 2025-02-11:
SE modbus 2.0.2
HA 
Core 2025.2.1
Operating System 14.2

adam-80, 2025-02-12:
I can confirm the same issue. It appears the integration will no longer connect to the inverter. I have the same integration version but I am running 2025.2.2.

I have confirmed that I can connect to the inverter on port 1502 with QModMaster and the data is still being fed to the Solar Edge cloud monitoring.

blowk, 2025-02-12:
I changed back to v1.13.0 for now.

PabloFFF93, 2025-02-12:
I went back to 1.0.13. And still the same problem.
I checked the settings of the Inverters. Each has a matching IP and each is set to port 502 (I turned off Modbus TCP and turned it back on). They respond to PING. They have access to the cloud. However, in NODERED they do not respond to Modbus - Read.
Answers 1 of 3

[image]

PabloFFF93, 2025-02-13:
I solved the problem. SOLAREDGE service (they did the optimizer mapping because I was adding panels) also changed the modbus ID settings. There is no mention of the modbus ID in the setup application. That's why I didn't know what was going on.

closed 2025-02-13 by PabloFFF93