binsentsu/home-assistant-solaredge-modbus #281: Battery energy dashboard not working since Solaredge Modbus 2.0

quinten94b, 2025-02-19:
Since update the Solaredge Modbus 2.0 integration the energy dashboard is not registering the battery consumption anymore.

I receive the following errors on my energy dashboard setup page:
•	sensore.solaredge_battery1_discharge and sensor.solaredge_battery1_charged are not in the expected device class (energy_storage)
•	sensore.solaredge_battery1_discharge and sensor.solaredge_battery1_charged have a state class ‘measurement’ but ‘last_reset’ is missing.

[image]

It happened the moment I upgraded from 1.13 to 2.0.

Best regards,
Quinten

sk72, 2025-02-23:
Hi Quinten, the last few days things are quite unstable here, next to the fact that some entities seem to be no longer functioning. After reloading the integration it works for a while (and show wrong data for today). 
I've read already some reports on HA modbus issues since a recent update and I think they might be connected.

quinten94b, 2025-02-23:
@sk72 My guess is that this particula issue is more related to the 2.0.0 update of this integration, which changed the state_class of some entities. Because of this, they do not fit the required state_class for the HA energy dashboard.

Jueff, 2025-02-24:
@binsentsu : 

in commit https://github.com/binsentsu/home-assistant-solaredge-modbus/commit/7be6b0d3920231e8e9bf90ebe6057b2f447708cf device class of battery charged/dischared has been changed from 'energy' to 'energy_storage'

follwing the comments in https://github.com/home-assistant/core/issues/135195 
device class 'energy_storage' is wrong.

see this comment
[kristian](https://github.com/kristian) [on Jan 9](https://github.com/home-assistant/core/issues/135195#issuecomment-2580350585)

Judging from https://github.com/home-assistant/core/blob/dev/homeassistant/components/energy/validate.py#L22 only "energy" is allowed as a device class. By my judgement of [device_classes allowed for sensors](https://www.home-assistant.io/integrations/sensor/#device-class), I would argue, especially for batteries, energy_storage should be added as a valid device_class for battries.

Jueff, 2025-02-24:
also changing the "state_class" from "total_increasing" to "measurement" breaks the funtionality.
sensor battery#_discharged and battery#_charged are no longer usable in HA energy configuration.

binsentsu, 2025-02-24:
@Jueff Thanks for analysis. In V2.0.3 the device class was changed to ENERGY and state class to TOTAL_INCREASING for these battery sensors.