binsentsu/home-assistant-solaredge-modbus #284: incompatibility: HA 2025.2 pymodbus 3.8.x; solaredge_modbus pymodbus 3.7.4; solardge_modbus_multi pymodbus 3.8.3

rud-ale, 2025-02-25:
Hey, i have following problem and one question:

I am using HA 2025.2 -> pymodbus 3.8.x

Custom integration(s):
solaredge_modbus ( 2.0.3 ) -> pymodbus 3.7.4
solaredge_modbus_multi ( 3.1.0 ) -> pymodbus 3.8.3 ( Updates for HA 2025.2 )

My Question:
Is there any planned update for HA 2025.2 / pymodbus 3.8.3+ ?
Or is there a newer ( beta ) version for HA 2025.2 using pymodbus 3.8.3+ ?

I am asking because of this:
I ran into incompatibilities between the 2 integrations using different versions of pymodbus...

 "WillCodeForCats" ( solaredge_modbus_multi ) wrote me:
My release v3.1.0 was updated for Home Assistant 2025.2 and pymodbus 3.8.x.
Uninstall or delete the solaredge_modbus integration from HA and restart should fix the problem.
2025-02-25 18:52:02.695 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] pymodbus version 3.7.4 

-> It he was right  - stopping "solaredge_modbus" fixed the problem.
[https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/764](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/764)

AdamWeglarz, 2025-03-13:
Solaredge's Modbus [manifest.json](https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/6a3179cb86063ba92a4457a31ce76ea128d251e7/custom_components/solaredge_modbus/manifest.json#L5) currently contains pymodbus==3.7.4 so you can first try to change it to pymodbus>=3.7.4.

rud-ale, 2025-03-13:
@AdamWeglarz : Thank you for you information.
I had exchanged my solaredge module to another one to get my automation working and ready for future releases of HA. Tthat's why I can't try it anymore.

But i will keep it in mind - as there can be same problem with other modules forceing use of different versions of pymodbus.

Is definitely worth a try.
Thank you for your help!

membersound, 2025-03-13:
@rud-ale which addon do you use now? same problem here.

rud-ale, 2025-03-13:
@membersound : i am using solaredge_modbus_multi currently. 
Do not know if it is better, but for me it is working.

binsentsu, 2025-04-08:
V2.0.4 was released with a soft pin of version iso hard pin. This should fix it if this pymodbus version dependency is running behind of HA core (https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/289)

closed 2025-04-08 by binsentsu

sffbigmac, 2025-04-09:
However if you are on a newer homeassistant version, which I have just now moving to 2025.4.1, coming with a newer 3.8.something pymodbus version the current integration code breaks:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solaredge_modbus/select.py", line 81, in async_select_option
    self.hub.write_registers(unit=self.hub._address, address=self._register, payload=new_mode)
    ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 299, in write_registers
    return self._client.write_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, values=payload, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 514, in write_registers
    return self.execute(no_response_expected, pdu_reg.WriteMultipleRegistersRequest(address=address, registers=values,dev_id=slave))
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/pdu.py", line 36, in __init__
    self.count: int = count or len(self.registers)
                               ~~~^^^^^^^^^^^^^^^^
TypeError: object of type 'int' has no len()
```

Also see this:
https://github.com/home-assistant/core/issues/138443#issuecomment-2675094282

Any idea by when this will be fixed? @binsentsu

sffbigmac, 2025-04-09:
@binsentsu here is a PR quickfix which did seem to work for me, not thoughrougly tested though:
https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/296
You might want to double check if that makes sense, also see my PR comments regarding backwards compatibility.

Aside from this I also started to see timeouts from data reads since the recent update of HA and solaredge modbus this morning (unrelated to that one liner PR above).
```
Logger: custom_components.solaredge_modbus
Source: helpers/update_coordinator.py:412
integration: SolarEdge Modbus ([documentation](https://github.com/binsentsu/home-assistant-solaredge-modbus))
First occurred: 14:22:55 (1 occurrences)
Last logged: 14:22:55

Error fetching solaredge data: Error updating modbus data: Modbus Error: [Connection] ModbusTcpClient yourhost:1502: Connection unexpectedly closed 0.000 seconds into read of unbounded read bytes without response from slave before it closed connection
```
As I didn't see this happening before I guess there must also be another issue around with the version upgrades and connection handling. Thoughts anyone?

sffbigmac, 2025-04-09:
@binsentsu for the record, I created a separate issue #297 for that recently reported connection error.